moggieuk/Happy-Hare #425: Retract after loading

Romano26, 2024-09-16:
Hello, I have a problem while loading the tool. Once the head is changed, ercf makes a big retraction while the filament is loaded in the extruder. The ercf pulls on the filament while it is in the extruder. Can you help me?
https://photos.app.goo.gl/2BQUFZuEFu7UYppr7

moggieuk, 2024-09-17:
It's not apparent what the problem is.  It appears as though the filament is not loaded, thus the first part of loading is to load the filament through the MMU gate.  This involves gripping the filament and extruding about 100mm so it is past and can be seen by the encoder.

On the ERCF design the gear with "jiggle" on servo down to aid engagement of the BMG gear. That can be defeated with `servo_buzz_gear_on_down` setting but it is recommended and standard on ERCF.

moggieuk, 2024-10-02:
Closing due to inactivity.

closed 2024-10-02 by moggieuk