moggieuk/Happy-Hare #426: [Request] Backwards Compatibility with Klipper 11

f00d4tehg0dz, 2024-09-16:
I am requesting to have the toolhead backwards compatible with Klipper 11. I and others cannot upgrade to Klipper 12 due to hardware constraints with our Flashforge Adventurer 5M series. 

The last commit that seems to work with Klipper 11 is https://github.com/moggieuk/Happy-Hare/tree/5d84cb25b83c494f0224929000f45e2a46678e21?tab=readme-ov-file#---installation  

However it lacks the new features and many bug fixes.

f00d4tehg0dz, 2024-09-16:
Disregard. I see you may have fixed it here: https://github.com/moggieuk/Happy-Hare/pull/388/commits

moggieuk, 2024-09-17:
I think this is addressed and PR388 is incorporated into the new v2.7.1 release.

Early access is available with:
```
./install.sh -b v271beta
```

I will point out that I cannot support earlier version of klipper myself because of the impracticality of installing an older system (not just klippy but re-flashing all the mcus).  Thus I will rely on community support to validate it works.

f00d4tehg0dz, 2024-09-17:
> I think this is addressed and PR388 is incorporated into the new v2.7.1 release.
> 
> Early access is available with:
> 
> ```
> ./install.sh -b v271beta
Thank you, and Happy-Hare works now with that beta branch. Thank you so much and the community!

closed 2024-09-17 by f00d4tehg0dz