moggieuk/Happy-Hare #429: Filament Changing Problem

Car1derWa1, 2024-09-17:
Hi there. 
First of all, exceptional project and sorry for my bad English.

I have a problem with my new ERCF v2. 
It is connected via CAN bus and completely set up (second time because of the problems).

Manually changing the filament with the t0/t1.... commands works really well (I never had a problem with that).
But during printing, the filament unloads and the print stops with the error "Reason: Unload sequence failed: It may be time to get the pliers out! Filament appears to be stuck somewhere. Occured when changing tool: T0 > T1".
But there is no stuck filament, I can type T1 in the console, it just switches the slot an loads the filament, then i can restarts the print and it works again until the next time. This happens pretty regularly every 5 to 10 layers. It happens in both slots

I have already tried to get some help on the Voron discord but they couldn´t help me until now. I think it is a software bug.
down here is my config @
[config-20240918-000026.zip](https://github.com/user-attachments/files/17035472/config-20240918-000026.zip)

Already thanks for your help

moggieuk, 2024-09-19:
This is an encoder validation test that is run after the unload.  Basically when the servo releases the filament, the encoder is asked if it sees movement.  If it does, HH concludes that the filament must still be through the encoder (how else could the encoder move?).

This test could fail if vibrations move the encoder wheel.  It could also fail if the encoder is not reliable.  The first problem is suppose to be prevented by Binky having a little brake inside -- when the filament is removed.  It may be technically possible that klipper is delaying the sending of encoder movement but I haven't experienced this.

I would first look closely at encoder .. is the brake working? does it spin on it's own?  Does the servo cause excessive vibrations?

You can turn off this validation check in s/w by setting:
```
encoder_move_validation: 0
```

Car1derWa1, 2024-09-22:
Thanks for the help found a very free spinningencoder wheel ( ill print a new one) and found out that the binky probe was defektive and not sensoring sometimes (two new are one the way)

Car1derWa1, 2024-10-01:
hi i am having problems again with my encoder (after spending 20 more hours in it). i printed a new encoder wheel and got a new binky encoder, the encoder works but sometimes it only reads half the distance. i disasambeld it and reasambeld it and it works again i have no idea whats going on and im close to just abandoning the encoder and replacing it with a simple micro switch. Anyone got any ideas?

closed 2024-11-14 by github-actions[bot]