moggieuk/Happy-Hare #430: Random suggestion: filament color validation in MMU_CHECK_GATE

k1-801, 2024-09-18:
HappyHare receives filament colors that a displayed in the UI.
A reasonable continuation would be adding filament color validation for the gates.
There are arduino-compadible color sensors (TCS3200 and alike) that can be integrated into the selector (somewhere after the encoder), which can provide a reading on the filament color, that can be compared to the color provided by the slicer. The reading obviously won't be accurate, so a tolerance (possibly, a WIDE tolerance) is necessary.

It is an absolutely random idea, but, theoretically, this kind of check can save some headache if the loading order is incorrect.

moggieuk, 2024-09-19:
Sounds like a wonderful and fun project....  but not something I have the energy for at the moment.  Note that there is a new "automap" function that you can set to be color.  So long as the gate map is accurate, then HH can automap the slicer tools to the correct gate...

moggieuk, 2026-10-21:
Closing this one because v4 now covers the idea with the TD-1 filament scanner rather than a TCS3200-type sensor.

In v4.1, if you enable "Has TD-1 scanner(s)?" in menuconfig, Happy Hare measures each gate's filament colour (and TD) as it passes the scanner:
- `MMU_CHECK_GATE TD1=1` measures any gates that haven't been measured yet (`TD1_UPDATE=1` re-measures all of them).
- `td1_auto_update: 1` applies new readings automatically, and `td1_capture_on_load: 1` takes a reading on a normal load.
- The measured colour becomes the gate's filament colour if none is set. `MMU_TD1 GATES=0,1 SET_COLOR=1` forces it.

Choose the `closest_color` automap strategy ("Automap strategy" in menuconfig, or `variable_automap_strategy` in `mmu_macro_vars.cfg`). HH will then map the slicer's tools to the gates whose measured colour is closest. That strategy only compares gates with the same material as the slicer tool, so the gate's material needs to be set too, either from Spoolman or with `MMU_GATE_MAP GATE=n MATERIAL=PLA`. In practice that catches (and fixes) spools loaded in the wrong order. It doesn't raise a "colour mismatch" alarm as such. NFC/RFID tags with Spoolman are another way to get the same result.

If there's something v4 still doesn't do for you here, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk