moggieuk/Happy-Hare #433: [request] Spoolman load temperature usage

hypersmc, 2024-09-21:
Simple request to have happy hare (the software) load the hotend temperature from spoolman of selected gate (that is linked to the spoolid) when loading the filament (outside of printing) This will make us able to have temperatures mentioned on spoolman and have it load them when we load the correct gate.

moggieuk, 2024-10-14:
So this would require:
- the filament temp to be added to the "gate map" and new mmu_vars variable.
  - MMU_GATE_MAP command updated
- Spoolman integration updated to send temp
- logic to ensure extruder temp = max(gate_map_temp, temp_min) when out of print

moggieuk, 2024-10-14:
This is now implemented in the v3.0.0 codebase.

This will be released soon to include support for type-B MMU's (AMS style).

moggieuk, 2026-10-21:
Closing this as done. It went in with v3.0 and is still in v4.

With Spoolman enabled (`spoolman_support` in `mmu.cfg`, or "Select spoolman spool manager support" in menuconfig), Happy Hare pulls each spool's extruder temperature from Spoolman into the gate map. You can also set it by hand with `MMU_GATE_MAP GATE=n TEMP=xxx`. When you load outside a print and the extruder is cold, HH heats to that gate's temperature. While printing it still follows the slicer.

If there's something v4 still doesn't do for you, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk