moggieuk/Happy-Hare #436: 3D Chameleon support

Firas95k, 2024-09-21:
I have a modified 3D Chameleon filament changer system using directly connected stepper drivers to my main board, it is awesome that it is simple and reliable way to do multiple filaments.... Advantages/Differences is that it does not need a servo and the gear stepper direction is reversed for T2 & T3 (T1&T2 is the same direction as the extruder), would be really awesome if it was supported.

Firas95k, 2024-09-22:
[gcode_macro CHAMELEON_HOME]
description: homes the tool selector.
variable_ysplit_extruder: 1120
variable_chameleon_ysplit: 100
variable_chameleon_speed: 300
variable_chameleon_acc: 100
variable_extrude_amount: 120
variable_chameleon_is_homed: 0
gcode:
  {% set svv = printer.save_variables.variables %}
  # Initialize or update the chameleon parameter
  {% if 'chameleon' not in svv %}
    {% set chameleon = [0, -1, 'empty', 'empty', 'empty', 'empty', -1] %}
  {% else %}
    {% set chameleon = svv.chameleon %}
    {% if chameleon[0] != -1 %}
      {% set chameleon = [chameleon[0]] + [chameleon[0]] + chameleon[2:] %}   #set previous=current
    {% else %}
      {% set chameleon = [chameleon[0]] + chameleon[1:] %}             #preserve previous state if current=-1 
    {% endif %}
  {% endif %}
  MANUAL_STEPPER STEPPER=selector SET_POSITION=0 
  MANUAL_STEPPER STEPPER=selector MOVE=-2.65 STOP_ON_ENDSTOP=1
  MANUAL_STEPPER STEPPER=selector SET_POSITION=0 
  MANUAL_STEPPER STEPPER=selector MOVE=0.12
  MANUAL_STEPPER STEPPER=selector SET_POSITION=0 
  # Set current tool to 0 after homing
  {% set chameleon = [0] + chameleon[1:] %}
  SAVE_VARIABLE VARIABLE=chameleon VALUE="{chameleon}"
  SET_GCODE_VARIABLE MACRO=CHAMELEON_HOME VARIABLE=chameleon_is_homed VALUE=1
  M118 Chameleon is Homed!
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro CHAMELEON_TS0]
gcode:
  _TOOL_SELECT TOOL=0
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro CHAMELEON_TS1]
gcode:
  _TOOL_SELECT TOOL=1
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro CHAMELEON_TS2]
gcode:
  _TOOL_SELECT TOOL=2
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro CHAMELEON_TS3]
gcode:
  _TOOL_SELECT TOOL=3
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro _TOOL_SELECT]
gcode:
  {% set TOOL = params.TOOL|int %}
  {% if printer["gcode_macro CHAMELEON_HOME"].chameleon_is_homed != 1 or printer.stepper_enable.steppers['manual_stepper selector'] != True %}
    M118 Chameleon is not homed, must run CHAMELEON_HOME
  {% else %}
    {% set svv = printer.save_variables.variables %}
    {% set chameleon = svv.chameleon %}
    {% set chameleon = [TOOL, chameleon[0]] + chameleon[2:] %}
    {% if TOOL == 0 %}
      MANUAL_STEPPER stepper=selector MOVE=0
    {% elif TOOL == 1 %}
      MANUAL_STEPPER stepper=selector MOVE=0.5
    {% elif TOOL == 2 %}
      MANUAL_STEPPER stepper=selector MOVE=1.0
    {% elif TOOL == 3 %}
      MANUAL_STEPPER stepper=selector MOVE=1.5
[... 107 lines cut ...]
  {% else %}
    {% if printer["gcode_macro CHAMELEON_HOME"].chameleon_is_homed != 1 or printer.stepper_enable.steppers['manual_stepper selector'] != True %}
      M118 Chameleon is not homed, must run CHAMELEON_HOME
    {% else %}
      # Update the tool status to 'soft'
      {% set chameleon = chameleon[:TOOL+2] + ['soft'] + chameleon[TOOL+3:6] + [chameleon[6]] %}
      SAVE_VARIABLE VARIABLE=chameleon VALUE="{chameleon}"
      {% set MOVE_DISTANCE = YSPLIT_DIST if TOOL == 0 or TOOL == 1 else -YSPLIT_DIST %}
      FORCE_MOVE STEPPER="extruder_stepper chameleon" DISTANCE={MOVE_DISTANCE} VELOCITY={SPEED} ACCEL={ACC}
      M118 Load tool {TOOL} to Y-splitter by {YSPLIT_DIST}mm and set status to 'soft' for tool {TOOL}
    {% endif %}
  {% endif %}
  CHAMELEON_CHK_ACTIVE_TOOL

[gcode_macro CHAMELEON_CHK_ACTIVE_TOOL]
gcode:
  {% set svv = printer.save_variables.variables %}
  {% set chameleon = svv.chameleon %}
  RESPOND TYPE=command MSG='Current active tool is tool {chameleon[0]}'
  RESPOND TYPE=command MSG='Previous active tool was tool {chameleon[1]}'
  RESPOND TYPE=command MSG='Tool status:'
  {% for tool in range(4) %}
    RESPOND TYPE=command MSG='Tool {tool}: {chameleon[tool+2]}'
  {% endfor %}
  RESPOND TYPE=command MSG='Tool loaded to extruder: {chameleon[6]}'

moggieuk, 2024-09-24:
There is a user that modified HH to work with Chameleon (or something like it).  It requires getting in pretty deep with the code to do that kind of change.

I'm currently have prototype multi-stepper support (I call in type-B mmu) but I don't have Chameleon or sponsorship so I'm unlikely to get to this any time soon..

u3dreal, 2024-10-11:
I get an error downloading the macros .. can you reupload please ?
I also hacked my version to work via a second  controller via clipper.
Would be interesting to compare.

Firas95k, 2024-11-14:
[Chameleon Macros.txt](https://github.com/user-attachments/files/17747518/Chameleon.Macros.txt)

Here it is

u3dreal, 2024-11-14:
Thanks alot. I am implementing it in Happay Hare right now. If you want to test
[https://github.com/u3dreal/Happy-Hare/tree/v3beta](url)

Firas95k, 2024-11-15:
> Thanks alot. I am implementing it in Happay Hare right now. If you want to test
> [https://github.com/u3dreal/Happy-Hare/tree/v3beta](url)

Your efforts are tremendous! Thank you✨ I would be happy to try it, is there a specific Readme or instructions for 3d Chameleon?

closed 2024-12-29 by github-actions[bot]