moggieuk/Happy-Hare #437: Gate 0 is empty!

danyial, 2024-09-22:
I receive this message every time I switch to T0 (except for the first load), and the print pauses. After I enter MMU_GATE_MAP GATE=0 AVAILABLE=1 in the console and resume the print, it continues until it switches back to T0 again.

Why is this happening?

[mmu-2.log](https://github.com/user-attachments/files/17089093/mmu-2.log)

moggieuk, 2024-09-24:
It's because you are getting a runout event on gate 0:
```
 > __MMU_GATE_RUNOUT GATE=0
 ```
 This is probably because the pre-gate sensor on gate 0 fired when T1 was loaded so filament is marked unavailable.
 
 You say this happens every time but I only see it once in the log.
 
 Possibly the pre-gate sensor is misbehaving on that gate.

danyial, 2024-09-24:
But the selector sits at Gate 1 at the runout, not at gate 0.

moggieuk, 2024-09-30:
I looked at the log you shared again.  This looks like a straight forward case of the gate having no filament:
```
DEBUG: Loading tool T0...
11:57:02 Gate 0 is empty!
         Use 'MMU_CHECK_GATE GATE=0' or 'MMU_GATE_MAP GATE=0 AVAILABLE=1' to reset.
         Occured when changing tool: T1 > T0. Retrying...
```
If gate 1 is selected that either you have a TTG map set to map T0 to gate 1, or the selector failed to move for mechanical problems.

moggieuk, 2024-10-02:
In v2.7.2 I make sure that the selector gate is at the correct location before issuing this error to avoid confusion. Marking as fixed.

closed 2024-10-02 by moggieuk