moggieuk/Happy-Hare #440: Request: Enable clog detection during user post load extension

CompyENG, 2024-09-24:
Would it be possible to enable clog detection during the user post load extension?

I'm running ERCFv1.1 with filametrix and [Ejection Assisted Purge](https://github.com/bombela/EjectionAssistedPurge) .  To make this "similar to blobifier", I put my PURGE macro as the user post load extension (since docs suggest to make this the BLOBIFIER command for blobifier).

I've been fighting random clogs during cutting as I tune things, but it seems like clog detection is *not* enabled during user post load extension.  So, this time of extrusion 'during change' is unable to detect a problem.

If a cut introduced a clog which caused the load to 'partially' fail, this won't be detected until printing, and I'm already missing extrusion.  If the area of the given color is small enough, it's possible that the clog won't be detected until *next* swap.  That's a shame, since I just had a nice big extrusion that could have detected the clog.

Generally, I wonder if purging via user post load extension should more be treated as 'in-print' rather than 'out-of-print'.  Of course, I'm still trying to get the clogs to simply not happen in the first place :upside_down_face:

moggieuk, 2024-09-24:
It is very complicated to add in a reliable way and it takes a huge amount of testing.  The problem is that you can't act on the "trigger" immediately... so have to let the move queue exhaust itself.  The problem with purging moves is that they are long.  No matter what the macro (POST_LAOD_MACRO)  will finish before any clog detection will fire.  That isn't much different from detecting it as the print resumes.

So I really don't know how to gracefully support this.  You could try hacking the blobifer macro so that it periodically checks the movement of the encoder (that's available via printer variables .. see wiki).  But note that klipper's macro implementation means that printer variables don't change within a macro so you have to construct one macro that calls others in a loop so the checking can take place..

CompyENG, 2024-09-25:
Thanks for the feedback!

Got it... I don't know anything about klipper internals, but it sounds like klipper maintains a move queue of 'upcoming' moves/extrusions and clog detection has to act somewhat 'behind' that.  Since the purge is a single move, the normal print will be queued behind it anyway and there really isn't a way for purge detection to act 'faster'.

I had assumed that putting the purge in my slicer toolchange instead of happy hare would have caught this clog, but it sounds like that's not the case anyway, so simply enabling clog detection for user post load won't actually solve anything.

Trying to 'hack' around this with a macro seems like an interesting idea... I'll have to play around with that.

Thanks for the feedback, and all the work you've done on Happy Hare!

closed 2024-09-25 by CompyENG