moggieuk/Happy-Hare #444: Problems with Restore Position since update to v2.7.1

weuzor, 2024-09-26:
Hi,
I am running happy hare with a custom park (user_pre_unload_extension), cut (form_tip_macro) and purge/prime/return (user_post_unload_extension) macro. The latter „steals“ the nextpos coordinates from the happy hare park macro and moves the toolhead where it needs to be to continue the print after doing its stuff.
This worked fine for the last couple of months but since the latest update happy hare would move the toolhead back to where the tool change started making a dot in the wrong color and  return to the correct (next) position before continuing the print, although the _MMU_RESTORE_POSITION macro would not suggest to do so.
When returning after parking it is the same except the toolhead returns to where it was parked, do some z move and return to where it already was.
(I have variable_enable_park_printing , variable_enable_park_standalone  and variable_enable_park_disabled  empty, so happy hare parking should be disabled as it used to be pre 2.7.1.)
Is there anything I can do to stop this from happening?

moggieuk, 2024-09-28:
Ok, so you have turned off parking correctly -- setting those three params will do that (and I assume you haven't set any of the other parking extensions).  The only other thing is that Happy Hare itself provides an overall safe guard to prevent printing in the wrong location. Without it folks were blaming for crashing into the bed for example.  This fail-safe move uses `restore_xy_pos` to decide what to do...

Either ensure the toolhead is at last position, or next position if available, or the current x,y position if set to `none`.

If you can read the python, search for the comment:
```
# Paranoia: no matter what macros do ensure position and state is good. Either last, next or none (current x,y)
```
Let me know if setting this variable doesn't allow you to do what you need.  I think it should (at least that was my intention).

weuzor, 2024-09-29:
Thanks for your reply. 

>(and I assume you haven't set any of the other parking extensions)
I am not sure which other parking extensions you mean but my relevant settings are the following:

Movement section:
variable_enable_park_printing 		: ''				; Empty '' to disable parking
variable_enable_park_standalone 	: ''				; Empty '' to disable parking
variable_enable_park_disabled  	: ''				; Empty '' to disable parking
variable_min_toolchange_z       		: 0.0				; The absolute minimum saftey floor (z-height) for ALL parking moves
variable_park_toolchange 			: -1, -1, 0, 0, 0		; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)
variable_park_runout				: -1, -1, 0, 0, 0		; x,y,z-hop,z_hop_ramp,retract
variable_park_pause				: -1, -1, 0, 0, 0		; x,y,z-hop,z_hop_ramp,retract (park position when mmu error occurs)
variable_park_cancel				: -1, -1, 0, 0, 0		; x,y,z-hop,z_hop_ramp,retract
variable_park_complete			: -1, -1, 0, 0, 0		; x,y,z-hop,z_hop_ramp,retract
variable_pre_unload_position  		: -1, -1, 0			; x,y,z-hop position before unloading starts
variable_post_form_tip_position 	: -1, -1, 0			; x,y,z-hop position after form/cut tip on unload
variable_pre_load_position      		: -1, -1, 0			; x,y,z-hop position before loading starts
variable_restore_xy_pos         		: "next"			; last|next|none - What x,y position the toolhead should travel to after a "toolchange"
variable_user_park_move_macro   	: ''				; Executed instead of default 'G1 X Y move' to park position

Extensions:
variable_user_pre_unload_extension   	: '_PURGINATOR_HAPPY_PRE_UNLOAD'		; Executed after default logic
form_tip_macro						: _PURGINATOR_HAPPY_FORM_TIP_MACRO
variable_user_post_form_tip_extension 	: '_PURGINATOR_HAPPY_POST_FORM_TIP'	; Executed after default logic
variable_user_post_unload_extension   	: '_PURGINATOR_HAPPY_POST_UNLOAD'	; Executed after default logic
variable_user_pre_load_extension      	: '_PURGINATOR_HAPPY_PRE_LOAD'		; Executed after default logic
variable_user_post_load_extension     	: '_PURGINATOR_HAPPY_POST_LOAD'		; Executed after default logic but before restoring toolhead position
variable_user_pause_extension   		: '_PURGINATOR_HAPPY_PAUSE'			; Executed after the klipper base pause
variable_user_resume_extension 		: '_PURGINATOR_HAPPY_RESUME'			; Executed before the klipper base resume
variable_user_cancel_extension  		: '_PURGINATOR_HAPPY_CANCEL'			; Executed before the klipper base cancel_print

> Either ensure the toolhead is at last position, or next position if available, or the current x,y position if set to none.
My user_post_load_extension macro moves the printhead to the next_pos stored in the happy hare park macro.
So after the user_post_load_extension ends the toolhead already is at happy hares next_pos correlating with variable_restore_xy_pos.

However, after that, the happy hare still would:
-After initial tool change move the toolhead from print start position (next_pos) back to start (park) position and once again to print start position (next_pos).
-After all other tool changes move from next pos to the position where the tool change started and the back to next pos.
-When resuming after pause, in which my user_resume_extension  uses its own stored coordinates for returning, the toolhead moves back to park pos and once again to the print. I wonder where happy hare would get that park pos coordinates from, as my parking position is nowhere configured in happy hare.

> Let me know if setting this variable doesn't allow you to do what you need. I think it should (at least that was my intention).
Unfortunately not :-(

moggieuk, 2024-09-29:
Hmmm, you really are fighting some pretty challenging logic here because there are so many combination of events.  However, I think one missing piece is that you need to call:
```
_MMU_CLEAR_POSITION
```
perhaps in your `post_load` macro.  That will ensure that when `_MMU_RESTORE_POSITION` is called it will do nothing.

You will next `restore_xy` to `next` or `none` otherwise the failsafe in HH itself will right you.

On the pause case, Happy Hare will doctor the saved `PAUSE_STATE` so that the base RESUME will always return to the expected position. This is necessary to prevent back-and-forth movement on resume.  Could that be the missing info you are looking for?

weuzor, 2024-09-29:
Thank you for this tip which sounds very promising. Unfortunately I cannot give it a try at the moment as my mmu servo died. Will report the outcome after replacing the servo.

moggieuk, 2024-09-29:
Ok.

I wondered, given the extensive parking and movement logic that now exists why you would need to handle all the movement in extension macros?

weuzor, 2024-09-29:
Curiosity made me change the servo in record time and I am happy to report putting _MMU_CLEAR_POSITION at the end of my return macro was the decisive hint and solved my issues for the tool changes. Thanks a lot!

However, resume after park is more resilient and still shows the described behavior although my user_resume_extension calls the very same return macro with the _MMU_CLEAR_POSITION command at its end.
And the happy hare client resume macro (which I use) also calls _MMU_RESTORE_POSITION which then should do nothing.
Even putting _MMU_CLEAR_POSITION in the user_pause_extension does not change that.
Is there another mechanic besides _MMU_RESTORE_POSITION which moves the toolhead after user_resume_extension?

> I wondered, given the extensive parking and movement logic that now exists why you would need to handle all the movement in extension macros?

Because I run these macros since times when happy hare did not have this functionality. They are very much optimized for the purge station I built.
We talked about it briefly quite a while ago if you remember. It changed a lot since then. If you want to have a look: github.com/weuzor/purginator
But it is definitely on my to do list to look into the new functions of happy hare more deeply and see if I can offload stuff to happy hare :-)

moggieuk, 2024-09-30:
Oh, I remember you know.  Sorry, just so many conversations....
The purginator is pretty cool - I need to take another look at it.

So, a little more detail on movement.  The `_MMU_CLEAR_POSITION` does reset the macros complete (as you have discovered) which is where almost all the logic now it.  It does not change what HH does internally and there are two things to note:
1. As described, HH has built in protection and will restore toolhead position no matter what. It does respect the `restore_xy_pos` setting. So while it will always restore the z-height, the x,y restoration can be last/next/none.  You will see a log message "Ensuring toolhead position..." when this logic runs.
2. HH manipulates the toolhead postions that is recorded by the BASE klipper pause. This is to prevent return to the wrong place on restore ... nothing else I can do to prevent BASE resume from being called because user has ultimate control on the RESUME macro.   So since I can intercept the initial call to RESUME, I inject what HH believes is the correct position into the "PAUSE_STATE" saved state recorded by the macro.  Thus the BASE resume will go where I want it to.  This will work as expected for "last" and "next" because the coordinates are known.  But I suspect will not work for "none".  What I probably need to do (and its an oversight) is, in the case of a "none" setting, inject the current x,y and the desired z.

I will look into the "none" handling for restore if you can verify the above makes sense in your scenario.

weuzor, 2024-09-30:
To completely rule out any involvement of _MMU_RESTORE_POSITION I changed it temporarily to

[gcode_macro _MMU_RESTORE_POSITION]
description: Restore saved toolhead position
gcode:
    _MMU_CLEAR_POSITION

.. which also had no effect so I assumed it must have to do with the built in protection.
However, setting  restore_xy_pos to „none“ also has no effect so I set it back to next :-(

I do have entries in my log:
20:37:59   DEBUG: Ensuring correct gcode state and position (X:48.9 Y:300.5 Z:0.4 E:191.7) after toolchange
after which all is good, toolhead is not moved

20:39:33   DEBUG: Ensuring correct gcode state and position (X:86.0 Y:301.5 Z:0.4 E:411.6) after resume
after which the toohead still moves from X:86.0 Y:301.5 Z:0.4 back to park position producing an impressive string and then move again to X:86.0 Y:301.5 Z:0.4 to smear it onto the print.

As mentioned before the park location the toolhead moves to is nowhere configured in happy hare so it must be stored somehow after the user_pause_extension and before the user_resume_extension.
Looking at the log entries the only difference I see is the „mmu pause wrapper“ and the „mmu resume wrapper“
Could the unwanted move be associated with one of them? 
Sorry to bother you so much with this.

Log:
Pause/resume
19:39:28   DEBUG: MMU PAUSE wrapper called
19:39:28   DEBUG: Saving toolhead gcode state and position (X:55.5 Y:266.9 Z:0.4 E:51.2) for pause
19:39:28 > mmu_sync_gear_motor
19:39:35 > RESUME
19:39:35   DEBUG: MMU RESUME wrapper called
19:39:35 > mmu_sync_gear_motor
19:39:39 > mmu_sync_gear_motor
19:39:40   DEBUG: Continuing from printing state after resume
19:39:42   DEBUG: Ensuring correct gcode state and position (X:55.5 Y:266.9 Z:0.4 E:468.8) after resume

Toolchange
20:42:58 > MMU_CHANGE_TOOL TOOL=3 NEXT_POS="48.903,300.503" ; T3
20:42:58   DEBUG: Tool change initiated with standalone MMU tip forming
20:42:58 Tool change requested, from T1 to T3
20:42:58   DEBUG: Saving toolhead gcode state and position (X:65.0 Y:308.9 Z:0.8 E:25.7) for toolchange
20:42:58 Restoring MMU gear stepper run current to 100% configured
20:43:00 Unsynced MMU from extruder
20:43:00   DEBUG: Unloading tool T1
20:43:00 > mmu_sync_gear_motor
20:43:00 Modifying MMU gear stepper run current to 70% for extruder syncing
20:43:00 Synced MMU to extruder
20:43:03 Unloading filament...
20:43:03 [T1] < En <<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
20:43:04 Restoring MMU gear stepper run current to 100% configured
20:43:04   DEBUG: Setting pressure advance for tip forming: 0.000000
20:43:04 Forming tip...
20:43:08   DEBUG: Restoring pressure advance: 0.000000
20:43:08 [T1] < En <<<<<<<<< (*) << [Ex << (*) <. Nz] -7.0mm (e:6.6mm)
20:43:09   DEBUG: Extracting filament from extruder
20:43:09   DEBUG: Reverse homing up to 83.0mm to extruder sensor (synced) to exit extruder
20:43:09 Unsynced MMU from extruder
20:43:12 [T1] < En <<<<<<<<<|(e) .. [Ex .. (t) .. Nz] -54.1mm (e:53.1mm)
20:43:12   DEBUG: Filament should be out of extruder
20:43:12   DEBUG: Unloading bowden tube
20:43:12   DEBUG: Performing bowden pre-unload test
20:43:13 [T1] < En <<<<<.... (e) .. [Ex .. (t) .. Nz] -95.0mm (e:67.3mm)
20:43:20 [T1] < En <........ (e) .. [Ex .. (t) .. Nz] -1078.1mm (e:1051.2mm)
20:43:20   DEBUG: Slow unload of the encoder
20:43:22 Warning: Possible encoder malfunction (free-spinning) during final filament parking
20:43:22 [T1] < En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED -1128.6mm (e:1101.4mm)
20:43:22   DEBUG: Setting servo to up (filament released) position at angle: 30
20:43:23 Unload of -1128.6mm filament successful (encoder measured 1101.4mm)
20:43:23   DEBUG: Autotune: bowden move ratio: 1.000799, Extra homing movement: 0.0mm
20:43:23   DEBUG: Loading tool T3...
20:43:24   DEBUG: Selecting tool T3 on Gate 3...
20:43:24   DEBUG: Setting servo to move (filament hold) position at angle: 30
20:43:24 Tool T3 enabled
20:43:24 Loading filament...
20:43:24 [T3] > En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
20:43:24   DEBUG: Setting servo to down (filament drive) position at angle: 140
20:43:25 > MMU_STATS COUNTER=servo_down INCR=1
20:43:27 [T3] > En >........ (e) .. [Ex .. (t) .. Nz] 79.2mm (e:79.6mm)
20:43:27   DEBUG: Loading bowden tube
20:43:34 [T3] > En >>>>>>>>> (e) .. [Ex .. (t) .. Nz] 1059.1mm (e:1061.6mm)
20:43:34   DEBUG: Loading filament into extruder
20:43:34   DEBUG: Homing up to 40.0mm to toolhead sensor (synced)
20:43:37 [T3] > En >>>>>>>>> (*) >> [Ex >>|(*) .. Nz] 1085.8mm (e:1088.2mm)
20:43:37   DEBUG: Loading last 55.0mm to the nozzle...
20:43:40 [T3] > En >>>>>>>>> (*) >> [Ex >> (*) >> Nz] LOADED 1140.8mm (e:1143.1mm)
20:43:40   DEBUG: Filament should loaded to nozzle
20:43:40 Load of 1140.8mm filament successful (encoder measured 1143.1mm)
20:43:40   DEBUG: Autotune: bowden move ratio: 1.002181, Extra homing movement: 16.7mm
20:43:40 > MMU_SLICER_TOOL_MAP PURGE_MAP=1
20:43:40 > mmu_sync_gear_motor
20:43:40 Modifying MMU gear stepper run current to 70% for extruder syncing
20:43:40 Synced MMU to extruder
20:44:04 > mmu_sync_gear_motor
20:44:06 MMU Statistics:
20:44:06   DEBUG: Continuing from printing state after toolchange
20:44:06   DEBUG: Ensuring correct gcode state and position (X:48.9 Y:300.5 Z:0.8 E:598.9) after toolchange
20:44:06 > __MMU_EXTRUDER_RUNOUT

moggieuk, 2024-10-02:
Ah, I think I get it.  My logic for the "Ensuring correct gcode state and position..." is:

If not paused:
```py
 # Paranoia: no matter what macros do ensure position and state is good. Either last, next or none (current x,y)
sequence_vars_macro = self.printer.lookup_object("gcode_macro _MMU_SEQUENCE_VARS", None)
 if sequence_vars_macro and sequence_vars_macro.variables.get('restore_xy_pos', 'last') == 'none' and self.saved_toolhead_operation in ['toolchange']:
```
(so only applies if we are running "MMU_CHANGE_TOOL" or "Tx" command.

If paused, then the logic is:
```py
# Prevent BASE_RESUME from moving toolhead
if self.TOOLHEAD_POSITION_STATE in self.gcode_move.saved_states:
                gcode_pos = self.gcode_move.get_status(self.reactor.monotonic())['gcode_position']
                self.gcode_move.saved_states['PAUSE_STATE']['last_position'][:3] = gcode_pos[:3]
```
So I THINK I need to added the "none" handling case to the resume wrapper logic.  Or doctor the resume pos with current x,y before allowing the resume wrapper to run.

moggieuk, 2024-10-02:
I know I tweaked this logic so first can you double check this still occurs with the latest Happy Hare.  I pushed my v2.7.2 work live now to main.

If you still have this problem I will fix, but I can't see how it is happening with current logic.

weuzor, 2024-10-02:
> I know I tweaked this logic so first can you double check this still occurs with the latest Happy Hare.
> I pushed my v2.7.2 work to main now. If you still have this problem I will fix, but I can't see how it is happening with current logic.

I updated to Happy Hare v2.7.2-20-gd70ff209 but the unwanted move still persists.

> The resume wrapper does this before calling your resume:
> # Prevent BASE_RESUME from moving toolhead
> if self.TOOLHEAD_POSITION_STATE in self.gcode_move.saved_states:
>     gcode_pos = self.gcode_move.get_status(self.reactor.monotonic())['gcode_position']
>     self.gcode_move.saved_states['PAUSE_STATE']['last_position'][:3] = gcode_pos[:3]

Unfortunately I don’t speak python but If the resume wrapper does this before my return macro is called
the toolhead is at that time at the position the unwanted back and forth move is directed to (y, y and z).

> What is your use case for pause / resume not restoring to "last" position?
I am not quite sure what you mean. After pause/resume my restore macro moves the toolhead to the position which my park macro saved before moving the toolhead.
So I guess this is „last“ position.

> Also, can you share your "RESTORE" macro

Hereafter are pause and resume extensions and the return macro. All other macros are also on the GitHub.

[gcode_macro _PURGINATOR_HAPPY_PAUSE]
description: Happy Hare User Pause Extension
gcode:
    _PURGINATOR_PARK

[gcode_macro _PURGINATOR_HAPPY_RESUME]
description: Happy Hare User Resume Extension
gcode:
    ## purge if a return position is set (prevents purging and errors if the print is paused before actual printing began)
    {% if printer['gcode_macro _PURGINATOR_RETURN'].return_pos_stored %}
        ## execute purge sequence, optionally retracting filament and descending gantry during servo waittimes
        _PURGINATOR_PURGE RETRACTSPEED={printer['gcode_macro _PURGINATOR_VARS'].return_retract_speed } DESCENDSPEED={printer['gcode_macro _PURGINATOR_VARS'].return_descend_speed}
        ## return to print suppressing the use of happy hare nextpos
        _PURGINATOR_RETURN NEXTPOS=0
    {% endif %}

[gcode_macro _PURGINATOR_RETURN]
description:  Return toolhead to print
variable_return_pos_stored:    False
variable_return_pos_x:         0
variable_return_pos_y:         0
variable_return_pos_z:         2
variable_return_travel_height: 2
gcode:
    ## respond macroname incl. parameters for debugging
    {% if printer['gcode_macro _PURGINATOR_VARS'].show_issued_commands %}
        RESPOND TYPE=command MSG="_PURGINATOR_RETURN {rawparams}"
    {% endif %}
    ## set parameters
    {% set printing = params.PRINTING | default ( printer.mmu.print_state == 'printing' ) %}
    {% set retractspeed = printer['gcode_macro _PURGINATOR_VARS'].return_retract_speed | float%}
    {% set descendspeed = printer['gcode_macro _PURGINATOR_VARS'].return_descend_speed | float %}
    {% set initial_layer_print_height = printer['gcode_macro _PURGINATOR_SLICER_VARS'].initial_layer_print_height | float %}
    ## sync gear motor
    mmu_sync_gear_motor
    ## return to print if return pos is stored
    {% if return_pos_stored and printing %}
        ## update return pos to happy hare next_pos if available and not prohibited by parameter NEXTPOS=0 (for resume after pause)
        {% if printer['gcode_macro _MMU_PARK'].next_pos and params.NEXTPOS | default ( 1 ) | int == 1 %}
            ## get next pos from happy hare park macro
            {% set nx,ny = printer['gcode_macro _MMU_PARK'].next_xy|map('float') %}
            {% set return_pos_x = nx %}
            {% set return_pos_y = ny %}
            ## respond begin of return sequence
            {% if params.RESPOND|default(printer['gcode_macro _PURGINATOR_VARS'].respond_detail_level)|int >= 1 %}
                RESPOND TYPE=command MSG="Returning toolhead to happy hare nextpos x: "{return_pos_x | round(2)}" y: "{return_pos_y | round(2)}" z: "{return_pos_z | round(2)}" ..."
            {% endif %}
        {% else %}
[... 14 lines cut ...]
            {% set zigzags = printer['gcode_macro _PURGINATOR_VARS'].return_brush_zigzags|int %}
            {% set start_y_depth = printer['gcode_macro _PURGINATOR_VARS'].return_brush_start_y_depth|int %}
            _PURGINATOR_BRUSH_NOZZLE WIPES={wipes} ZIGZAGS={zigzags} DEPTH={start_y_depth} RETRACTSPEED={retractspeed} DESCENDSPEED={descendspeed}
        {% endif  %}
        ## travel to park_z_hop above return position, optionally retracting filament
        _purginator_travel X={return_pos_x} Y={return_pos_y} Z={return_travel_height} RETRACTSPEED={retractspeed}
        ## lower nozzle to print level, optionally retracting filament
        _purginator_travel Z={return_pos_z} RETRACTSPEED={retractspeed}
        ## unretract filament
        _purginator_filament_unretract SPEED={printer['gcode_macro _PURGINATOR_VARS'].return_unretract_speed}
        ## clear return position
        _purginator_clear_return_position
        ## clear mmu position to prevent happy hare from moving the toolhead beyond this point
        _MMU_CLEAR_POSITION
        ## respond end of return sequence
        {% if params.RESPOND|default(printer['gcode_macro _PURGINATOR_VARS'].respond_detail_level)|int >= 1 %}
            RESPOND TYPE=command MSG="Toolhead returned"
        {% endif %}   
    ## not printing or no return position 
    {% else %}
        ## undock with nozzle wipe
        _PURGINATOR_UNDOCK WIPE=True {rawparams}
        ## unretract any accumulated retraction
        _purginator_filament_unretract SPEED={printer['gcode_macro _PURGINATOR_VARS'].return_unretract_speed}
    {% endif %}

weuzor, 2024-10-02:
> I know I tweaked this logic so first can you double check this still occurs with the latest Happy Hare.
> I pushed my v2.7.2 work to main now. If you still have this problem I will fix, but I can't see how it is happening with current logic.

I updated to Happy Hare v2.7.2-20-gd70ff209 but the unwanted move still persists.

> The resume wrapper does this before calling your resume:
> # Prevent BASE_RESUME from moving toolhead
> if self.TOOLHEAD_POSITION_STATE in self.gcode_move.saved_states:
>     gcode_pos = [...]

Unfortunately I don’t speak python but If the resume wrapper does this before my return macro is called
the toolhead is at that time at the position the unwanted back and forth move is directed to (y, y and z).

> What is your use case for pause / resume not restoring to "last" position?
I am not quite sure what you mean. After pause/resume my restore macro moves the toolhead to the position which my park macro saved before moving the toolhead.
So I guess this is „last“ position.

> Also, can you share your "RESTORE" macro

Hereafter are pause and resume extensions and the return macro. All other macros are also on the GitHub.

[gcode_macro _PURGINATOR_HAPPY_PAUSE]
description: Happy Hare User Pause Extension
gcode:
    _PURGINATOR_PARK

[gcode_macro _PURGINATOR_HAPPY_RESUME]
description: Happy Hare User Resume Extension
gcode:
    ## purge if a return position is set (prevents purging and errors if the print is paused before actual printing began)
    {% if printer['gcode_macro _PURGINATOR_RETURN'].return_pos_stored %}
        ## execute purge sequence, optionally retracting filament and descending gantry during servo waittimes
        _PURGINATOR_PURGE RETRACTSPEED={printer['gcode_macro _PURGINATOR_VARS'].return_retract_speed } DESCENDSPEED={printer['gcode_macro _PURGINATOR_VARS'].return_descend_speed}
        ## return to print suppressing the use of happy hare nextpos
        _PURGINATOR_RETURN NEXTPOS=0
    {% endif %}

[gcode_macro _PURGINATOR_RETURN]
description:  Return toolhead to print
variable_return_pos_stored:    False
variable_return_pos_x:         0
variable_return_pos_y:         0
variable_return_pos_z:         2
variable_return_travel_height: 2
gcode:
    ## respond macroname incl. parameters for debugging
    {% if printer['gcode_macro _PURGINATOR_VARS'].show_issued_commands %}
        RESPOND TYPE=command MSG="_PURGINATOR_RETURN {rawparams}"
    {% endif %}
    ## set parameters
    {% set printing = params.PRINTING | default ( printer.mmu.print_state == 'printing' ) %}
    {% set retractspeed = printer['gcode_macro _PURGINATOR_VARS'].return_retract_speed | float%}
    {% set descendspeed = printer['gcode_macro _PURGINATOR_VARS'].return_descend_speed | float %}
    {% set initial_layer_print_height = printer['gcode_macro _PURGINATOR_SLICER_VARS'].initial_layer_print_height | float %}
    ## sync gear motor
    mmu_sync_gear_motor
    ## return to print if return pos is stored
    {% if return_pos_stored and printing %}
        ## update return pos to happy hare next_pos if available and not prohibited by parameter NEXTPOS=0 (for resume after pause)
        {% if printer['gcode_macro _MMU_PARK'].next_pos and params.NEXTPOS | default ( 1 ) | int == 1 %}
            ## get next pos from happy hare park macro
            {% set nx,ny = printer['gcode_macro _MMU_PARK'].next_xy|map('float') %}
            {% set return_pos_x = nx %}
            {% set return_pos_y = ny %}
            ## respond begin of return sequence
            {% if params.RESPOND|default(printer['gcode_macro _PURGINATOR_VARS'].respond_detail_level)|int >= 1 %}
                RESPOND TYPE=command MSG="Returning toolhead to happy hare nextpos x: "{return_pos_x | round(2)}" y: "{return_pos_y | round(2)}" z: "{return_pos_z | round(2)}" ..."
            {% endif %}
        {% else %}
            ## respond begin of return sequence
[... 13 lines cut ...]
            {% set zigzags = printer['gcode_macro _PURGINATOR_VARS'].return_brush_zigzags|int %}
            {% set start_y_depth = printer['gcode_macro _PURGINATOR_VARS'].return_brush_start_y_depth|int %}
            _PURGINATOR_BRUSH_NOZZLE WIPES={wipes} ZIGZAGS={zigzags} DEPTH={start_y_depth} RETRACTSPEED={retractspeed} DESCENDSPEED={descendspeed}
        {% endif  %}
        ## travel to park_z_hop above return position, optionally retracting filament
        _purginator_travel X={return_pos_x} Y={return_pos_y} Z={return_travel_height} RETRACTSPEED={retractspeed}
        ## lower nozzle to print level, optionally retracting filament
        _purginator_travel Z={return_pos_z} RETRACTSPEED={retractspeed}
        ## unretract filament
        _purginator_filament_unretract SPEED={printer['gcode_macro _PURGINATOR_VARS'].return_unretract_speed}
        ## clear return position
        _purginator_clear_return_position
        ## clear mmu position to prevent happy hare from moving the toolhead beyond this point
        _MMU_CLEAR_POSITION
        ## respond end of return sequence
        {% if params.RESPOND|default(printer['gcode_macro _PURGINATOR_VARS'].respond_detail_level)|int >= 1 %}
            RESPOND TYPE=command MSG="Toolhead returned"
        {% endif %}   
    ## not printing or no return position 
    {% else %}
        ## undock with nozzle wipe
        _PURGINATOR_UNDOCK WIPE=True {rawparams}
        ## unretract any accumulated retraction
        _purginator_filament_unretract SPEED={printer['gcode_macro _PURGINATOR_VARS'].return_unretract_speed}
    {% endif %}

weuzor, 2024-11-13:
showing life sign to prevent going stale

TTom1088, 2024-11-16:
I have the same problem, but at first I suspected a configuration error on my part. Then I came across this report.
In my case the error is a bit meaner, as it can destroy the entire printer. In my case the printer moves to the last corner after the break and then to Z4.5 to get to the brush, after cleaning it moves up to the previously saved Z coordinate and only then starts printing again.
If an error occurs during cleaning (e.g. low extruder temperature)
The happy hare then moves directly to the print, diagonally in the Z coordinate, so the gantry then moves against the print. Normally it would simply do nothing and I could intervene and then press resume again.

I have of course also done my "research" to prevent pointless back and forth travel. I have found an "unclean solution" for this. I simply commented out my own "RESTORE_GCODE_STATE" in my resume macro.

weuzor, 2024-12-17:
activity to prevent stale

weuzor, 2024-12-17:
showing activity to prevent stale status.

weuzor, 2024-12-17:
reopen

weuzor, 2024-12-17:
problem still there

weuzor, 2025-01-18:
Showing activity against staleness  and hoping for a fix.

weuzor, 2025-01-18:
trying to reopen

dooodking, 2025-01-27:
Hello everyone. I have Voron 2.4 ERCF, Filametrix and Blobifier. Happy Hare 3.0. My problem is that variable_restore_xy_pos : "next" does not work, it always uses the LAST position, where should I look for an answer to this problem, it always puts a dot of a different color on the part. 

[image]

closed 2025-03-12 by github-actions[bot]