moggieuk/Happy-Hare #447: MMU_TEST_CONFIG Do Not work

Thomsen1982, 2024-09-28:
When i use the command MMU_TEST_CONFIG, the adapted settings are Not loaded

moggieuk, 2024-09-28:
What do you mean "not loaded".  They will not be saved \, you have to do that manually.  The idea is that you can try changes to see if they are what you want but they only remain active until the next restart

Thomsen1982, 2024-09-29:
I thought if I change something in the config and then save it. Everything during printing. Then I call MMU_TEST_CONFIG. This applies the settings for the current print. The output in the console then shows the new values. That's how I understood it, is that wrong?
Do I have to call MMU_TEST_CONFIG with the parameter and the value?

Thomsen1982, 2024-09-29:
Oh i read the Wiki again and i thing i used it wrong. Call MMU_TEST_CONFIG alone only show me the possible parametersi can change. Call MMU_TEST_CONFIG with parameter change the value for the print

closed 2024-09-29 by Thomsen1982