moggieuk/Happy-Hare #448: Problem with blobbs on cleaning Tower

Thomsen1982, 2024-09-28:
With an earlier version it was perfect adjusted.
But with current version i have big blobbs on cleaning tower when loading the filament.
It is Not possible to compensate it.
Change the parameter toolhead_ooze_reduction has no effect.
Also Adaption in the cutting macro has no effect.

moggieuk, 2024-09-28:
[image]

There was actually a bug in the logic prior that had little effect with small ooze_reduction but did when it got larger.  Therefore it is possible that if you optimized before it might have been by luck rather than "science".  I needed to close this bug because it is impossible to help folks otherwise.  That said it is possible the maths is still wrong, although I've reviewed it several times.

The picture is from a spreadsheet I have that represents the maths involved and what parameters are significant. If you can provide your parameters I can enter them and see what we expect in terms of loading distance. BTW that final loading distance will be in the log messages as:
```
Loading last 43.5mm to the nozzle...
```
It is at the DEBUG level so you will likely have to look in `mmu.log` to find it.

So, give my all your parameters to put into this sheet and check the log message...

moggieuk, 2024-09-29:
I've identified an issue when using tip cutting that completely explains your experience.  I will be released v2.7.2 shortly to address.

Thomsen1982, 2024-09-30:
Hi,

I don't have time to test it right now. But a colleague was able to take a look.
This is what he says about the update:

So from my point of view it still doesn't work with the blob setting. I have now reduced extruder to nozzle and sensor to nozzle by 10mm each, now it works.
If I only increase the new parameter residual filament to 50 mm and the old ooze to 25 mm there is no improvement

Thomsen1982, 2024-10-03:
Now I've had time to test the fix.

I don't know exactly what my colleague tested, but it works for me.
I only got my cleaning tower free through the parameter toolhead_ooze_reduction = 10.

good work

closed 2024-10-03 by Thomsen1982