moggieuk/Happy-Hare #449: Restart Printer nozzle is heating

Thomsen1982, 2024-09-29:
It is Not clean why.
The last mmu Print is finished without problems.
All end macros are set correct. Before there are no problems.
Start the printer or make a restart the mmu heats to the default temperature.
I used additional mmu end and mmu reset but nothing helped.

moggieuk, 2024-09-29:
I'm sorry, but I can't understand what the problem is.  Also, you need to attached `mmu.log` along with description of problem.

Thomsen1982, 2024-09-29:
The problem is, when I switch on the printer, the mmu immediately reports that the nozzle is being heated. Likewise with Save and restart. Then an unload is attempted. This happens again and again, as if the mmu has not finished printing properly and is trying to catch up. Is such information stored anywhere? I will attach the log.
[mmu.log](https://github.com/user-attachments/files/17180009/mmu.log)

moggieuk, 2024-09-30:
Ok, the `mmu.log` made it clear.  The "state" of the filament in the MMU is not known so it attempts to recover position at startup.  HOWEVER, it should bypass the test that requires the nozzle to heat up.  I will fix in the v2.7.2 release which is due very soon.

JFYI. If you shutdown with filament in a known position (like fully unloaded or fully loaded) this will not occur.

moggieuk, 2024-09-30:
This was a good catch and although it is not easier to get into the state you are, HH should not heat on startup.  I've fixed in the v2.7.2 release which is currently available in beta/early access.  See the announcement message below:

:rabbit: :rabbit::rabbit:   **Happy Hare - v2.7.2 **  :rabbit: :rabbit::rabbit: 

**WHY SO SOON?** Unfortunately I came to an impasse on how to address the catch-22 problem when cutting filament tips and also trying to tune loading length to avoid blobbing.

While everything works as expected when tip_forming, the use of `toolhead_ooze_reduction` on tip cutting would not alter the effective filament load length as designed (the reason was because of second use in calculating cut filament fragment).

To fix, I need to add an additional `mmu_parameter.cfg` and that requires re-running `./install.sh`, hence the need for new version.

To jump on the beta:
```
./install.sh -b v272beta
```
**WHAT TO EXPECT:** Upgrade is very straight forward and your previous value for `toolhead_ooze_reduction` will be moved to `toolhead_residual_filament`  (to better represent what it means) and ooze reduction setting will reset to 0 allowing you then to truly use it to control blobbing on the wipe tower in all situations. All doc in the wiki has been updated.

There are also a couple of other QoL improvements and hopefully more TTC mitigation: [Change Log](https://github.com/moggieuk/Happy-Hare/wiki/Change-Log#v272)

I don't expect thing to be long in beta but when you notice the branch has gone you need to run the `./install.sh -b main` again to jump back

moggieuk, 2024-10-02:
v272 is live. Closing issue as fixed.

closed 2024-10-02 by moggieuk