moggieuk/Happy-Hare #450: MMU_Servo in down position after RESUME

Stephan-83, 2024-09-29:
After pause and Resume with "variable_user_resume_extension  : 'BLOBIFIER'" Servo stay in down Position.
Pause and RESUME macro from client_macros.cfg

sync_to_extruder: 0
sync_form_tip: 1

[klippy.log](https://github.com/user-attachments/files/17179478/klippy.log)

moggieuk, 2024-09-29:
Which servo?

Stephan-83, 2024-09-29:
Yes the mmu_servo on the selector, blobifier for restart the nozzle (oozing in pause state, cooling heating, debuging).
mmu.log until tomorrow. sorry

moggieuk, 2024-09-30:
Firstly, BLOBIFER isn't designed to be used in the `variable_user_resume_extension`.  This is a very strange use case.  The blobifier marco is one that I ship with Happy Hare but not one I maintain.

In general on an mmu error (and pause) gear/extruder syncing will be disabled and the servo raised.  It is then restored on resume if in print.  If blobifer macro is called outside of a print it would be up to that macro to determine the final state of anything it controls.

I provide any more guidance you need to attached your `mmu.log` so I can see the sequence of events

Stephan-83, 2024-09-30:
Here is the MMU.Log
print start > pause via Klipperscreen > resume > filament runout > cancel print
Perhaps it is better not to use blobifier at this point, I'll think about another way.

thanks for your support
[mmu2024_09_30_1613.log](https://github.com/user-attachments/files/17191684/mmu2024_09_30_1613.log)

moggieuk, 2024-09-30:
In the log I see a call to:
`MMU_SYNC_GEAR_MOTOR SYNC=1`
Who is calling that?  It is not something Happy Hare does.  This is asking for the gear stepper to be synced with the extruder and doing so requires the servo to be down.

EDIT: I see that the blobifer macro is doing this (I'm actually not the author of this so didn't know).  
But is it reading your `sync_to_extruder` setting from `mmu_parameters.cfg` which appears to be set to "1", therefore blobifer will end with servo in the down position. However printing then starts which would have synced the gear and moved the servo anyway.

If you disable syncing:
```yml
sync_to_extruder: 0
```
The servo should remain up.

Stephan-83, 2024-10-01:
Issue found in blobifier:
{% if printer.configfile.config.mmu.sync_to_extruder%}
            MMU_SYNC_GEAR_MOTOR SYNC=1
          {% else %}
            MMU_SERVO POS=up
          {% endif %}
Always runs in MM_SYNC

have edited to {% if printer.configfile.config.mmu.sync_to_extruder == "1"%} working as expected

moggieuk, 2024-10-02:
Thanks. I update the blobifer macro in v272 release