moggieuk/Happy-Hare #451: Update to v2.7.1-59-g84ba4f32 Error

MDegelmann, 2024-09-29:
Warning: Error booting up MMU: 'RunoutHelper' object has no attribute 'enable_runout'

Printer Not work :(

moggieuk, 2024-09-30:
You haven't supplied any logs or config information, but I'm guessing that one of your MMU sensors is not configured correctly.  Have you defined all the mmu sensors: pre-gate, gate, extruder, toolhead in the `[mmu_sensors]` section?

moggieuk, 2024-09-30:
Assuming that I'm correct above. I've fixed this in the v2.7.2 release.  This is currently available as beta/early access.  See notice below:

:rabbit: :rabbit::rabbit:   **Happy Hare - v2.7.2 **  :rabbit: :rabbit::rabbit: 

**WHY SO SOON?** Unfortunately I came to an impasse on how to address the catch-22 problem when cutting filament tips and also trying to tune loading length to avoid blobbing.

While everything works as expected when tip_forming, the use of `toolhead_ooze_reduction` on tip cutting would not alter the effective filament load length as designed (the reason was because of second use in calculating cut filament fragment).

To fix, I need to add an additional `mmu_parameter.cfg` and that requires re-running `./install.sh`, hence the need for new version.

To jump on the beta:
```
./install.sh -b v272beta
```
**WHAT TO EXPECT:** Upgrade is very straight forward and your previous value for `toolhead_ooze_reduction` will be moved to `toolhead_residual_filament`  (to better represent what it means) and ooze reduction setting will reset to 0 allowing you then to truly use it to control blobbing on the wipe tower in all situations. All doc in the wiki has been updated.

There are also a couple of other QoL improvements and hopefully more TTC mitigation: [Change Log](https://github.com/moggieuk/Happy-Hare/wiki/Change-Log#v272)

I don't expect thing to be long in beta but when you notice the branch has gone you need to run the `./install.sh -b main` again to jump back

moggieuk, 2024-10-02:
v272 is now live. Closing issue

closed 2024-10-02 by moggieuk