moggieuk/Happy-Hare #452: PAUSE after toolchange

berreka, 2024-09-29:
After updating Happy Hare, my printer is pausing after every tool change. Running RESUME in the terminal works as expected, and the print continues, but it pauses again after the next tool change.

I tried setting:

variable_user_post_load_extension: 'RESUME'

But something still causes the print to pause after this. I'm unsure if I should go down the path of using delayed_gcode in mmu_sequence.cfg, or if there is something missing in my configuration

Here is mmu.log:
``` 20:42:54 > __MMU_BOOTUP
20:42:54 (\_/)
         ( *,*)
         (")_(") Happy Hare v2.7.1 Ready...
20:42:54   DEBUG: Job State: READY -> INITIALIZED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 43200.00s)
20:42:54   DEBUG: Setting servo to move (filament hold) position at angle: 20
20:42:54 > MMU_STATS COUNTER=mmu_restarts INCR=1
20:42:54 > MMU_STATS COUNTER=servo_down LIMIT=5000 WARNING="Inspect servo arm for wear/damage"
20:42:54 > MMU_STATS COUNTER=cutter_blade LIMIT=3000 WARNING="Inspect/replace filament cutting blade"
20:42:58   DEBUG: Job State: INITIALIZED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 43200.00s)
20:49:02 > MMU_CHANGE_TOOL TOOL=0
20:49:02 > MMU_PRINT_START
20:49:03 Happy Hare initialized ready for print (no filament preloaded)
20:49:03   DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 43200.00s)
20:49:03 MMU not homed, homing it before continuing...
20:49:03 Homing MMU...
20:49:03 Attempting to recover filament position...
20:49:03 [T?] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
20:49:03 Unloading filament...
20:49:03 [T?] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
20:49:04   DEBUG: Setting pressure advance for tip forming: 0.000000
20:49:04 Forming tip...
20:49:11   DEBUG: Restoring pressure advance: 0.090000
20:49:12 [T?] < En <<<<<<<<< [Ex << (*) <. Nz] -47.0mm (e:47.3mm)
20:49:12   DEBUG: Extracting filament from extruder
20:49:12   DEBUG: Setting servo to up (filament released) position at angle: 20
20:49:12   DEBUG: Reverse homing up to 71.0mm to toolhead sensor
20:49:14 [T?] < En <<<<<<<<< [Ex <<|(t) .. Nz] -65.7mm (e:65.4mm)
20:49:14   DEBUG: Unloading last 10.0mm to exit the extruder
20:49:15 [T?] < En <<<<<<<<< [Ex .. (t) .. Nz] -63.0mm (e:66.1mm)
20:49:15   DEBUG: Filament should be out of extruder
20:49:15   DEBUG: Unloading bowden tube
20:49:15   DEBUG: Setting servo to down (filament drive) position at angle: 110
20:49:15 > MMU_STATS COUNTER=servo_down INCR=1
20:49:15   DEBUG: Performing bowden pre-unload test
20:49:17 [T?] < En <<<<<.... [Ex .. (t) .. Nz] -78.0mm (e:84.9mm)
20:49:22 [T?] < En <........ [Ex .. (t) .. Nz] -670.3mm (e:684.9mm)
20:49:22   DEBUG: Slow unload of the encoder
20:49:27 [T?] < En ......... [Ex .. (t) .. Nz] UNLOADED -778.9mm (e:782.4mm)
20:49:27   DEBUG: Setting servo to up (filament released) position at angle: 20
20:49:28 Unload of -778.9mm filament successful (encoder measured 782.4mm)
20:49:28   DEBUG: Autotune: bowden move ratio: 1.013001, Extra homing movement: 45.6mm
20:49:28   DEBUG: Setting servo to move (filament hold) position at angle: 20
20:49:28   DEBUG: Setting servo to move (filament hold) position at angle: 20
20:49:29   DEBUG: Selecting tool T0 on Gate 0...
20:49:29   DEBUG: Setting servo to move (filament hold) position at angle: 20
20:49:29 Tool T0 enabled
20:49:29   DEBUG: Tool change initiated with slicer tip forming
20:49:29 Tool change requested: T0
20:49:29   DEBUG: Saving toolhead gcode state and position (X:155.0 Y:120.0 Z:10.0 E:0.0) for toolchange
20:49:30   DEBUG: Loading tool T0...
20:49:30 Loading filament...
20:49:30 [T0] > En ......... [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
20:49:30   DEBUG: Setting servo to down (filament drive) position at angle: 110
20:49:31 > MMU_STATS COUNTER=servo_down INCR=1
20:49:33 [T0] > En >........ [Ex .. (t) .. Nz] 72.2mm (e:72.4mm)
20:49:33   DEBUG: Loading bowden tube
20:49:38 [T0] > En >>>>>>>>> [Ex .. (t) .. Nz] 670.3mm (e:705.8mm)
20:49:38   DEBUG: Loading filament into extruder
20:49:38   DEBUG: Homing up to 800.0mm to toolhead sensor (synced)
20:49:40 [T0] > En >>>>>>>>> [Ex >>|(*) .. Nz] 694.9mm (e:731.6mm)
20:49:40   DEBUG: Loading last 61.0mm to the nozzle...
20:49:44 [T0] > En >>>>>>>>> [Ex >> (*) >> Nz] LOADED 755.9mm (e:792.1mm)
20:49:44   DEBUG: Filament should loaded to nozzle
20:49:44 Load of 755.9mm filament successful (encoder measured 792.1mm)
20:49:44   DEBUG: Autotune: bowden move ratio: 1.058976, Extra homing movement: 24.6mm
         Encoder measurement on gate 0 was outside of desired calibration range. You may want to check function or recalibrate
20:49:44   DEBUG: Spoolman spool_id not set for current gate
20:49:44 > RESUME
20:49:44   DEBUG: MMU RESUME wrapper called
20:49:44 Print is not paused. Resume ignored.
20:49:45   DEBUG: Setting servo to up (filament released) position at angle: 20
20:49:46 MMU Statistics:
         +-----------+-----------------------+------------------------+----------+
         |  1836(1)  |       unloading       |        loading         | complete |
         |   swaps   | pre  |    -    | post | pre  |    -    |  post |   swap   |
         +-----------+------+---------+------+------+---------+-------+----------+
         |    total  | 2.68 | 7:41:28 | 2.71 | 5.76 | 8:28:40 | 43.10 |  1:00:38 |
         |     └ avg | 0.00 |    15.1 | 0.00 | 0.00 |    16.6 |  0.02 |     1.98 |
         | this job  |    - |    24.6 |    - | 0.30 |    14.1 |  0.30 |     15.0 |
         |     └ avg |    - |    24.6 |    - | 0.30 |    14.1 |  0.30 |     15.0 |
         |      last |    - |    24.6 |    - | 0.30 |    14.1 |  0.30 |     15.0 |
         +-----------+------+---------+------+------+---------+-------+----------+
         
         7:15:23 spent paused over 78 pauses (All time)
         0.00 spent paused over 0 pauses (This job)
         Number of swaps since last incident: 8 (Record: 52)
         
         Gate Statistics:
         0:😎, 1:😎, 2:😎, 3:😊, 4:—, 5:—, 6:—, 7:—, 8:😃
20:49:46   DEBUG: Continuing from printing state after toolchange
20:49:46 Restoring toolhead position to (x:155.0, y:120.0, z:10.0)
20:49:46   DEBUG: Ensuring correct gcode state and position (X:155.0 Y:120.0 Z:10.0 E:-45.0) after toolchange
20:49:53 > MMU_PRINT_START
20:49:53 > PAUSE
20:49:53   DEBUG: MMU PAUSE wrapper called
20:49:53   DEBUG: Saving toolhead gcode state and position (X:100.0 Y:-3.0 Z:0.3 E:9.0) for pause
``` 
Any advice or suggestions would be greatly appreciated, thanks in advance!

moggieuk, 2024-09-30:
In the brief section of log you attached I'm not able to see the issue, but I do see that "PAUSE" and "RESUME" are being called directly -- Happy Hare never does this so I suspect something else in your configuration is doing so.  Do you have filament runout sensors setup perhaps?

berreka, 2024-09-30:
Yes, I have a filament runout sensor. In my mmu.cfg file, I had the following:

```
[filament_switch_sensor toolhead_sensor]
switch_pin = !TOOLHEAD_SENSOR
```
Adding
`pause_on_runout: False`

Seemed to solve the issue, but another pause just happened mid-print. I'm still investigating.

Here is the start of the klippy.log with all the configuration:
https://pastebin.com/QS13gLGn

And the mmu.log section
```
13:08:23 MMU Statistics:
         +-----------+-----------------------+-----------------------+----------+
         |  1863(23) |       unloading       |        loading        | complete |
         |   swaps   | pre  |    -    | post | pre  |    -    | post |   swap   |
         +-----------+------+---------+------+------+---------+------+----------+
         |    total  | 10.0 | 7:48:24 | 10.5 | 13.9 | 8:35:34 | 52.1 |  1:15:02 |
         |     └ avg | 0.00 |    15.1 | 0.01 | 0.01 |    16.6 | 0.03 |     2.42 |
         | this job  | 6.25 |    5:55 | 6.67 | 6.96 |    5:52 | 7.02 |    12:13 |
         |     └ avg | 0.27 |    15.5 | 0.29 | 0.30 |    15.3 | 0.30 |     31.9 |
         |      last | 0.27 |    15.7 | 0.30 | 0.30 |    14.1 | 0.31 |     32.1 |
         +-----------+------+---------+------+------+---------+------+----------+
         
         7:15:23 spent paused over 78 pauses (All time)
         0.00 spent paused over 0 pauses (This job)
         Number of swaps since last incident: 35 (Record: 52)
         
         Gate Statistics:
         0:😎, 1:😎, 2:😎, 3:😊, 4:—, 5:—, 6:—, 7:—, 8:😃
13:08:23   DEBUG: Continuing from printing state after toolchange
13:08:23 Restoring toolhead position to (x:220.1, y:127.9, z:4.4)
13:08:23   DEBUG: Ensuring correct gcode state and position (X:220.1 Y:127.9 Z:4.4 E:602.2) after toolchange
13:10:10 > MMU_CHANGE_TOOL TOOL=1
13:10:10   DEBUG: Tool change initiated with slicer tip forming
13:10:10 Tool change requested, from T0 to T1
13:10:10   DEBUG: Saving toolhead gcode state and position (X:179.9 Y:136.5 Z:4.6 E:-39.0) for toolchange
13:10:13   DEBUG: Unloading tool T0
13:10:13 Unloading filament...
13:10:13 [T0] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
13:10:13 Tip forming performed by slicer but 'slicer_tip_park_pos' not set
13:10:13   DEBUG: Extracting filament from extruder
13:10:13   DEBUG: Reverse homing up to 71.0mm to toolhead sensor
13:10:14 [T0] < En <<<<<<<<< [Ex <<|(t) .. Nz] -4.0mm (e:4.2mm)
13:10:14   DEBUG: Unloading last 10.0mm to exit the extruder
13:10:15 [T0] < En <<<<<<<<< [Ex .. (t) .. Nz] -63.0mm (e:5.6mm)
13:10:15   DEBUG: Filament should be out of extruder
13:10:15   DEBUG: Unloading bowden tube
13:10:15   DEBUG: Setting servo to down (filament drive) position at angle: 110
13:10:15 > MMU_STATS COUNTER=servo_down INCR=1
13:10:15   DEBUG: Performing bowden pre-unload test
13:10:17 [T0] < En <<<<<<<<< [Ex << (t) .. Nz] -78.0mm (e:13.2mm)
13:10:17   DEBUG: Setting servo to up (filament released) position at angle: 20
13:10:18 MMU issue detected. Print will be paused
         Reason: Unload sequence failed: Bowden pre-unload test failed. Filament seems to be stuck in the extruder or filament not loaded.
         Occured when changing tool: T0 > T1
13:10:18 After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
13:10:19   DEBUG: Job State: PRINTING -> PAUSE_LOCKED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 230.0, Resume to state: printing, Position saved for: pause, pause_resume: False, Idle timeout: 72000.00s)
13:10:19 > PAUSE
13:10:19   DEBUG: MMU PAUSE wrapper called
13:10:20 Attempting to recover filament position...
13:10:20   DEBUG: Checking for filament in MMU...
13:10:20   DEBUG: Setting servo to down (filament drive) position at angle: 110
13:10:20 > MMU_STATS COUNTER=servo_down INCR=1
13:10:22   DEBUG: Filament detected in encoder after buzzing gear motor
13:10:22 [T0] < En <<<<<.... [Ex .. (t) .. Nz] -78.0mm (e:17.4mm)
13:10:22   DEBUG: Setting servo to up (filament released) position at angle: 20
13:20:18 Disabled extruder heater
```

It seems like another macro is triggering the PAUSE. The curious thing is that the filament sensor wasn't stopping the print before. Thanks for pointing me in the right direction!

moggieuk, 2024-09-30:
You are almost right.  The best option would be to define the sensor in the
```
[mmu_sensors]
```
section in `mmu_hardware.cfg`
If you have an old hardware file (these are not automatically upgraded) then view the "template" version here:
```yml
~/Happy-Hare/config/base/mmu_hardware.cfg
```
You will see the correct way to define sensors for your MMU.  There are reasons for doing this and it is essential for pre-gate or extruder sensors.

closed 2024-10-02 by moggieuk