moggieuk/Happy-Hare #453: SET_LED LED=mmu_leds INDEX=0 RED=0 GREEN=0 BLUE=0 TRANSMIT=1

CopOsiris, 2024-09-30:
I get the error when starting the printer/MMU
Error running _MMU_SET_LED: Error on 'SET_LED LED=mmu_leds INDEX=0 RED=0 GREEN=0 BLUE=0 TRANSMIT=1': INDEX must have minimum of 1

If I do MMU_LED...
`LED animations: enabled
Default exit effect: 'gate_status'
Default entry effect: Unavailable
Default status effect: 'slicer_color'

Options:
ENABLE=[0|1]
ANIMATION=[0|1]
EXIT_EFFECT=[off|gate_status|filament_color|slicer_color]
ENTRY_EFFECT=[off|gate_status|filament_color|slicer_color]
STATUS_EFFECT=[off|on|filament_color|slicer_color]
12:11
Error running _MMU_SET_LED: Error on 'SET_LED LED=mmu_leds INDEX=0 RED=0 GREEN=0 BLUE=0 TRANSMIT=1': INDEX must have minimum of 1
12:11
Error on 'SET_LED LED=mmu_leds INDEX=0 RED=0 GREEN=0 BLUE=0 TRANSMIT=1': INDEX must have minimum of 1
12:11
Error on 'SET_LED LED=mmu_leds INDEX=0 RED=0 GREEN=0 BLUE=0 TRANSMIT=1': INDEX must have minimum of 1
12:11
MMU_LED`

moggieuk, 2024-09-30:
What is your led configuration in `mmu_hardware.cfg`?

CopOsiris, 2024-09-30:
happy-hare
v2.7.1-59-g84ba4f32
mainsail
v2.12.0
# (comment out this section if you don't have leds)
[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 6 #7	    # Number gates x1 or x2 + 1 (if you want status)
color_order: GRBW		# Set based on your particular neopixel specification

[mmu_leds]
num_gates: 6
led_strip: neopixel:mmu_leds
exit_range: 5-0 # 0-5
#entry_range: 7-12
status_index: 6
frame_rate: 24

moggieuk, 2024-09-30:
Ok. So chain_count should be 7 (because you have 6 exit and the status). I.e. 7 leds in chain. Then define as:
```
exit_range: 6-1
status_index: 7
```

CopOsiris, 2024-09-30:
The solution was
[mmu_leds]
num_gates: 6
led_strip: neopixel:mmu_leds
exit_range: 6-1
status_index: 0
frame_rate: 24

Putting this gave me an index error because it could not be greater than 6
exit_range: 6-1
status_index: 7

moggieuk, 2024-09-30:
So how many LED's do you actually have connected?

CopOsiris, 2024-10-01:
I'm only 6. I'm going to comment on the status_led...Tonight I'll tell you how it goes

closed 2024-11-14 by github-actions[bot]