moggieuk/Happy-Hare #456: Possible change to the BLOBIFIER gcode needed to resolve 'park_vars' is undefined

emericklaw, 2024-10-02:
I was getting the follwing error when setting up Blobifier.
>Error evaluating 'gcode_macro BLOBIFIER:gcode': jinja2.exceptions.UndefinedError: 'park_vars' is undefined

After reading all the discussions on the Discord channel relating to this error it turns out that I was not completing a XYZ home and QGL.

The macro does try and check the state of homing and QGL:

https://github.com/moggieuk/Happy-Hare/blob/d70ff2096d45b2564cacc0d23139b127cb2c1bc1/config/addons/blobifier.cfg#L277-L282

However, after debugging this, the macro errors before even attempting to run the checks.

Potential solution might be to have the BLOBIFIER macro only carry out the homing checks and all the actual BLOBIFIER logic in the `else` gets moved to a separate macro called within the `else`.

moggieuk, 2024-10-02:
I'm pretty sure this is fixed.  Are you up-to-date with HH?   (A v2.7.2 was just release which will require an upgrade: `./install.sh` but it's a simple process/change)

emericklaw, 2024-10-02:
Yep, pulling the latest 2.7.2 gracefully tells me I have not homed.

I was running v2.7.1-60 last night, I was up way too late tinkering so only logged the issues this afternoon.

Thanks!

closed 2024-10-02 by emericklaw