moggieuk/Happy-Hare #457: Happy Hare Install on Nebula pad Ender 3 v3 KE? is it possible?

SPYKE357, 2024-10-03:
Good morning all

Hope you are all well.
I sent comms to k1-801, one of the K1 contributors asking the same question and he advised i should post here to see if anyone has had any success and if so, how?
sorry to bug you all with this issue of mine but i was wondering if it's possible to install HappyHare for a ERCF v2 kit i bought ( still to build) on the ender3v3ke/nebula pad? i'd like to keep the nebula pad if possible as the only other mcu i have is a pi zero 2(also not sure if there is a full klipper release for that printer...
the ke/nebula pad has been rooted and i see the boards are similar/same as the k1. been trying the last couple of days to install it but no luck so far... do you know if it's possible? and if so how and are there any limitations?

if i do this command through ssh "uname -m" it returns "mips" as a response.

Any and all help will be massively appreciated.

SPYKE357, 2024-10-03:
-sh: ./install.sh: not found

this is returned when i run the ./install.sh line

jacksky6, 2024-10-03:
@SPYKE357  
You can take a look at the record of my installation process on ender3v3 plus for reference.
I put it on youtube.

Here is the link:
https://www.youtube.com/watch?v=B0I6bcV8F6Y

moggieuk, 2024-10-13:
Sorry, I have no way to help with this. But the "./install.sh not found" suggests you are not running from the Happy-Hare directory.  Did you:
```yml
cd ~/Happy-Hare
./install.sh
```?

closed 2024-11-28 by github-actions[bot]