moggieuk/Happy-Hare #459: ramming_volume_standalone value is always ignored in __MMU_FORM_TIP

skyguy94, 2024-10-03:
I've enabled forced standalone tip forming 
```python
force_form_tip_standalone: 1  # 0 = Slicer in print else standalone, 1 = Always standalone tip forming (TURN SLICER OFF!)
form_tip_macro: _MMU_FORM_TIP # Name of macro to call to perform the tip forming (or cutting) operation
```

and I have the following for ramming volumes configured.:
```python
variable_ramming_volume            : 0          ; Volume in mm^3, 0 = disabled (optionally let slicer do it)
variable_ramming_volume_standalone : 23         ; Volume in mm^3, 0 = disabled
```

If I run `MMU_TEST_FORM_TIP` then the standalone ramming value is not used (I don't see the oooze that comes out with ramming).  I believe this is because the standalone variable is initialized to the printing variable during initialize. 
```python
[gcode_macro _MMU_FORM_TIP]
description: Standalone macro that mimics Superslicer process

gcode:
...
    {% set ramming_volume = vars['ramming_volume']|float %}
    {% set ramming_volume_standalone = vars['ramming_volume']|float %}
...
```
The logic itself seems fine and setting the printer variable to the same value works around the issue.

```python
    # Step 1 - Ramming
    # This is very generic and unlike slicer does not incorporate moves on the wipetower
    {% set ramming_volume = ramming_volume_standalone if not printing else ramming_volume %}
    {% if ramming_volume > 0 %} # Standalone Ramming
        {% set ratio = ramming_volume / 23.0 %}
```

moggieuk, 2024-10-13:
Yes. That's a bug. You are correct that `ramming_volume_standalone` is set to `ramming_volume`.  I will push a patch..

moggieuk, 2024-10-13:
Fix pushed.  If for some reason there is still a problem, please re-open the issue.

closed 2024-10-15 by moggieuk

skyguy94, 2024-10-15:
LGTM. Thank you!