moggieuk/Happy-Hare #460: Unexpected result with selector touch capability

agreen, 2024-10-06:
**Test scenario:**

- ERCF v2
- Selector touch configured and enabled
- Selector physical endstop configured and enabled (NOT using virtual selector endstop)
- Filament blocking gate from moving
- MMU/printer reset
- Run MMU_HOME

**Expected result:**

- MMU attempts to home
- MMU realizes it cannot move
- MMU attempts to clear/unblock gate
- Gate unblocked
- MMU homes to physical endstop

**Actual result:**

- MMU attempts to home
- MMU "grinds" selector for a second or two
- MMU reports homing error "MMU issue: Homing selector failed because of blockage or malfunction. Klipper reports: No trigger on mmu_sel_home after full movement"

I have confirmed that the "touch" capability works AFTER homing when moving from one gate to another.  The documentation is not clear that the machine should be homed before this functionality starts working, and I had expected it to work prior to homing as well.

I would suggest either implementing this capability, or documenting this is only useful AFTER a successful MMU home.  Personally, the attempt of clearing the blockage on homing would be a big benefit, IMO.

moggieuk, 2024-10-13:
You are correct in your assessment - the "touch" automatic recovery only is applied to selector movement and not to homing.  There is a reason for this that would need to be carefully thought thru.. specifically, is the axis is not yet homed, normal movement is not possible.  It is possible to defeat this but it means bypassing all of klippers default homing logic. It is a good idea so I'll add to the "enhancement list".

moggieuk, 2024-10-13:
EDIT: I spent a couple of hours implementing this and then realized why I didn't do it before... (if selector is not aligned you cannot reliability unload).  Instead you can use this:
```
MMU_HOME FORCE_UNLOAD=1
```
This will ensure that filament is unloaded prior to the home.  Can't remember why I didn't make this the default (??)

moggieuk, 2024-10-13:
Also, can you confirm the state of the MMU after restart?  I.e. the output of `MMU_STATUS`.   I'm going to assume that it is in an UNLOADED state (because all other states will recover position and unload if necessary) and that a piece of filament has just poked through the gate to block the selector?

I ask because I'm trying to understand the exact case that is causing you problems.

agreen, 2024-10-13:
I'll do some further testing and get back. After I opened this bug, I have noticed sometimes it works as expected, sometimes it doesn't.  FYI, I've been chasing TTC errors as well ,and I suspect there's some correlation with the TTC errors.   I'm at about a 75% TTC to start print ratio. :-/

moggieuk, 2024-10-14:
Many folks have been having success with a new (fast) sd-card..

closed 2024-11-27 by github-actions[bot]