moggieuk/Happy-Hare #461: Fix compatibility of creality ender3v3 Plus

jacksky6, 2024-10-07:
Creality Ender3v3 Plus is used for hardware and systems similar to k1 series.

Using Happy Hare 2.7.2 firmware can solve most compatibility issues, except for the following:

In the init of ToolHead class of ender3v3 series, there is an attribute definition
``` 
class ToolHead:
    def __init__(self, config):
        ....
        self.gap_auto_comp = None
        ....
```

It will be used in the move method
```
    def move(self, newpos, speed):
        gap_ofts = [0., 0., 0., 0.]
        bed_ofts = [0., 0., 0., 0.]
        now_pos = self.get_position()
        self.gcode = self.printer.lookup_object('gcode')
        if self.gap_auto_comp != None:
            gap_ofts = self.gap_auto_comp.deal_move(self, now_pos, newpos)
            if self.gap_auto_comp.show_msg:
                self.gcode.respond_info('gap_auto_comp:X=%.3f, Y=%.3f, Z=%.3f' % (gap_ofts[0], gap_ofts[1], gap_ofts[2]))

        newpos = [sum(gb) for gb in zip(gap_ofts, bed_ofts, list(newpos))]

        # self.record_z_pos(newpos[2])
        move = Move(self, self.commanded_pos, newpos, speed)
        if not move.move_d:
            return
        if move.is_kinematic_move:
            self.kin.check_move(move)
        if move.axes_d[3]:
            self.extruder.check_move(move)
        self.commanded_pos[:] = move.end_pos
        self.move_queue.add_move(move)
        if self.print_time > self.need_check_stall:
            self._check_stall()
```

Without this attribute definition, the following error will be reported when moving.
[image]

I have not found a better feature to judge when this attribute needs to be added.

If you don't care too much, you can directly add the definition of this attribute in the init method.

or
```
import inspect

class MmuToolHead(toolhead.ToolHead, object):
    def __init__(self):
        ...
        source_code = inspect.getsource(toolhead.ToolHead.__init__)
        if 'gap_auto_comp' in source_code:
            self.gap_auto_comp = None
        ...
   
```
This is the `toolhead.py` of klipper on this ender3v3 plus.
[toolhead-ender3v3plus.zip](https://github.com/user-attachments/files/17280715/toolhead-ender3v3plus.zip)

Thank you!

moggieuk, 2024-10-13:
Adding an extra class variable is safe to do in the mainline branch.  Why not create a PR for this?

jacksky6, 2024-10-13:
@moggieuk 

Because I don't know whether to add this class variable directly or use the previous less ideal judgment method.

What do you think? Add it directly?

moggieuk, 2024-10-14:
From you reference move() def it looks like this is a customized version of klipper so this might get difficult fast.  If simply adding:
```
class MmuToolHead(toolhead.ToolHead, object):
    def __init__(self):
        self.gap_auto_comp = None # For Creality Ender3v3 Plus (custom klipper)
```

Is sufficient then this would be harmless to add to the init() method.

closed 2024-10-14 by jacksky6