moggieuk/Happy-Hare #465: Spoolman starts after ERCF causing failed connection

jwhite, 2024-10-14:
I was having an issue where my instance of Spoolman is running in a docker container.

Happy Hare will attempt to connect to the server before it is ready causing an error (and retrying the connection will work.)

I added this to my klipper systemd file in /etc/systemd/system/klipper.service to resolve the issue

(Note the additions of docker.service in the Unit section)

``
[Unit]
Description=Klipper 3D Printer Firmware SV1
Documentation=https://www.klipper3d.org/
After=network-online.target docker.service
BindsTo=docker.service
ReloadPropagatedFrom=docker.service
Wants=udev.target

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=pi
RemainAfterExit=yes
WorkingDirectory=/home/pi/klipper
EnvironmentFile=/home/pi/printer_data/systemd/klipper.env
ExecStart=/home/pi/klippy-env/bin/python $KLIPPER_ARGS
Restart=always
RestartSec=10
```

Maybe this could be added to the documentation?   I realise it is probably a Spoolman issue more than a Happy Hare issue but that is how it manifests.

moggieuk, 2024-10-15:
There really isn't anything I can do because it is impossible to distinguish between a slow starting spoolman and one that doesn't start. Also, many folks, myself included have spoolman on a different system. Happy Hare will retry when needed but you will see the first connection error.

I see what you are doing with the delayed start.  My fear is two fold: 1) this is very intimidating to most users, 2) the new more common way of installing spoolman is without docker (this starts up more quickly).

I'm happy to a section to the doc (https://github.com/moggieuk/Happy-Hare/wiki/Spoolman-Support) but can you write for me?  Also it should explain the general race condition, then offer this is a solution if running with docker.  Perhaps there is an equivalent setup for the non-docker install.

Actually, better still would be simply added a note to bottom of the Spoolman wiki page and linking to a new section in the "Troubleshooting" page (https://github.com/moggieuk/Happy-Hare/wiki/Troubleshooting-and-Common-Issues).  Then it can be less complete...

jwhite, 2024-10-16:
Yeah, happy to add a note.