moggieuk/Happy-Hare #466: Resume after clog occasionally returns to a position 10cm higher in Z

jwhite, 2024-10-16:
In some cases, such as after a clog or other print pause, the resume puts the toolhead in a position 10cm over the position it should be in.  I notice this is almost always before the purge block print (it is over the purge block.)

Here is the console:

```
15:25:35 
// Unsynced MMU from extruder
15:25:35 
// Unsynced MMU from extruder
15:25:35 
// Attempting to recover filament position...
15:25:36 
// [T3] < En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED -907.5mm (e:878.8mm)
15:25:36 
// Synced MMU to extruder
15:25:37 
// Unsynced MMU from extruder
15:27:39 
$ mmu_unlock
15:27:39 
// idle_timeout: Timeout set to 1800.00 s
15:27:39 
// Enabled extruder heater
15:27:44 
$ mmu_home
15:27:44 
// Homing MMU...
15:27:49 
// Tool T0 enabled
15:27:54 
$ MMU_CHECK_GATE GATE=3
15:27:54 
// Checking Gate 3...
15:27:59 
// Gate 3 - Filament detected. Marked available
15:28:03 
// Tool T0 enabled
15:28:03 
// Gates: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 |
// Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
// Avail: | B |   | S | S |   | S |   | B | B |
// Selct: | * |-------------------------------- T0
16:07:59 
$ MMU_CHECK_GATE GATE=3
16:07:59 
// Checking Gate 3...
16:08:04 
// Gate 3 - Filament detected. Marked available
16:08:09 
// Tool T0 enabled
16:08:09 
// Gates: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 |
// Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |
// Avail: | B |   | S | S |   | S |   | B | B |
// Selct: | * |-------------------------------- T0
16:08:22 
$ T3
16:08:22 
// Tool change requested: T3
16:08:22 
// Saving toolhead position (x:150.0, y:10.0, z:37.4)
16:08:22 
// Parking toolhead at (x:20.0, y:25.0, z:37.6) for toolchange operation
16:08:23 
// Tool T3 enabled
16:08:25 
// Loading filament...
16:08:25 
// [T3] > En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
16:08:28 
// [T3] > En >........ (e) .. [Ex .. (t) .. Nz] 69.9mm (e:69.8mm)
16:08:39 
// [T3] > En >>>>>>>>> (*) .. [Ex .. (t) .. Nz] 814.0mm (e:810.0mm)
16:08:39 
!! Warning: Automatically heating extruder to default temp (200.0°C)
16:08:39 
// Waiting for extruder to reach target (default) temperature: 200.0°C
16:11:05 
// [T3] > En >>>>>>>>> (*) >> [Ex >>|(*) .. Nz] 830.3mm (e:825.9mm)
16:11:08 
// [T3] > En >>>>>>>>> (*) >> [Ex >> (*) >> Nz] LOADED 872.6mm (e:867.8mm)
16:11:08 
// Load of 872.6mm filament successful (encoder measured 867.8mm)
16:11:24 
// MMU Statistics:
// +--------+------------------------------+----------------------------+----------+
// |  3873  |          unloading           |          loading           | complete |
// | swaps  |  pre  |       -       | post |  pre  |    -     |   post  |   swap   |
// +--------+-------+---------------+------+-------+----------+---------+----------+
// | total  | 15:27 | 8054175:48:31 | 7:03 | 30:39 | 17:08:29 | 6:49:03 | 36:47:34 |
// |  └ avg |  0.24 |    2079:34:13 | 0.11 |  0.48 |     15.9 |    6.34 |     34.2 |
// |   last |     - |             - |    - |  0.84 |     2:42 |    14.5 |     3:01 |
// +--------+-------+---------------+------+-------+----------+---------+----------+
// 
// 166:53:07 spent paused over 361 pauses (All time)
// Number of swaps since last incident: 1 (Record: 229)
// 
// Gate Statistics:
// 0:😎, 1:😃, 2:😎, 3:😎, 4:😎, 5:😎, 6:😎, 7:😎, 8:😎
16:17:10 
$ RESUME
16:17:10 
// Restoring toolhead position to (last) (x:150.0, y:10.0, z:37.4)
16:17:35 
// Synced MMU to extruder
16:17:35 
// Happy Hare initialized ready for print (initial tool T3 loaded)
16:17:35 
// Saving toolhead position (x:224.8, y:207.8, z:37.4)
16:17:35 
// Retracting 5.0mm
16:17:35 
// Parking toolhead at (x:n/a, y:n/a, z:47.4) for cancel operation
16:17:38 
// Unsynced MMU from extruder
```
​
Notice the final position is at z=47.4 not 37.4.

jwhite, 2024-10-16:
Full mmu log here:

```
16:17:09   DEBUG: MMU RESUME wrapper called
16:17:09   DEBUG: Continuing from idle state after resume
16:17:09 Restoring toolhead position to (last) (x:150.0, y:10.0, z:37.4)
16:17:09   DEBUG: Ensuring correct gcode state and position (X:150.0 Y:10.0 Z:37.8 E:11221.8) after resume
16:17:29 > CANCEL_PRINT
16:17:29 > MMU_PRINT_START
16:17:29   DEBUG: Job State: IDLE -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 1800.00s)
16:17:29   DEBUG: Setting servo to down (filament drive) position at angle: 43
16:17:32 > MMU_STATS COUNTER=servo_down INCR=1
16:17:34 Synced MMU to extruder
16:17:34 Happy Hare initialized ready for print (initial tool T3 loaded)
16:17:34   DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: True, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 1800.00s)
16:17:34   DEBUG: MMU_CANCEL_PRINT wrapper called
16:17:34   DEBUG: Saving toolhead gcode state and position (X:224.8 Y:207.8 Z:37.4 E:1924.4) for cancel
16:17:34 Saving toolhead position (x:224.8, y:207.8, z:37.4)
16:17:34 Retracting 5.0mm
16:17:34 Parking toolhead at (x:n/a, y:n/a, z:47.4) for cancel operation
16:17:35   DEBUG: Job State: PRINTING -> CANCELLED (MMU State: Encoder: Disabled, Synced: True, Paused temp: None, Resume to state: ready, Position saved for: , pause_resume: False, Idle timeout: 1800.00s)
16:17:35   DEBUG: Setting servo to up (filament released) position at angle: 138
16:17:37 Unsynced MMU from extruder

moggieuk, 2024-10-16:
The problem seems to be that the print cancel is kicking in and that parks +10mm higher.   Not quite sure why cancel has been called. If you didn't call yourself then perhaps on_error_gcode is set:
```
#on_error_gcode: CANCEL_PRINT
```
Look in `mmu_macro_vars.cfg` or elsewhere in your printer.cfg. 

If the on_error handler is kicking in there should be a gocde error that is causing it