moggieuk/Happy-Hare #468: Incomplete filament position recovery in setups with Extruder Entry sensor

smb209, 2024-10-18:
My MMU is configured with an extruder entry sensor but no toolhead sensor. I had several restarts where the filament was past the entry sensor but not yet engaged with the extruder gears. Despite the filament detected in the entry sensor, the EREC_CUTTER_ACTION (`variable_user_post_unload_extension`) was performed while the filament was still (partially) loaded.

moggieuk, 2024-10-25:
Unfortunately I can't help without an `mmu.log` file attached.  I assume that the MMU "state" is out of sync.  What is the status when in this situation?  Did you try "MMU_RECOVER", etc..

moggieuk, 2024-10-25:
I just saw your PR ... I will look at this, makes sense!

moggieuk, 2024-10-28:
Closing. PR pulled and incorporated.

Heads up: In v3.0.0 release (currently the v3alpha branch) I reworked the recovery logic. I'm pretty sure it will match your expectations but something to watch for when you get to that release...

closed 2024-10-28 by moggieuk