moggieuk/Happy-Hare #471: custom macros defined in `mmu_macro_vars.cfg` aren't kept on release upgrades

HitLuca, 2024-10-18:
I have seen for a bunch of version upgrades that any macro I have defined in `mmu_macro_vars.cfg` isn't kept, resulting in having to dig around to find their definitions on the backed up folders.
An example below:

```cfg
[...]
variable_user_post_load_extension: 'MY_POST_LOAD_EXTENSION' 

[gcode_macro MY_POST_LOAD_EXTENSION]
gcode:
    MMU_SYNC_GEAR_MOTOR  
    CLEAN_NOZZLE
```

Having the macro defined in this file makes sense, and having it defined right after I use it makes easier to keep everything under control.

Is there a suggested approach to solve this? Can we keep custom macros between upgrades? Happy to provide more details

wdcocq, 2024-10-21:
I'm not sure if it's documented anywhere. But I wouldn't advise putting it in `mmu_macro_vars.cfg`. These files get indeed overridden in an upgrade. 

So for now it's probably better to add it in something like `/mmu/mmu_custom_macros.cfg` (should not be touched by an upgrade, but would still be in your backup) and include that in `printer.cfg`

That said, I'm not the maintainer so maybe this will be supported in the future.

HitLuca, 2024-10-21:
I ended up moving my macro to `mmu.cfg` since that one gets copied when upgraded. That's where I also have a temperature sensor entry for the mmu board (which would be convenient to automatically add on first ercf install imo)

During upgrade I would then prefer to have a warning or automatic check that lets me know some functionalities won't be transferred, as in this case I would end up discovering it after the first color change of a (possibly) already long print

moggieuk, 2024-10-25:
To allow for upgrading functionality all the packaged cfg files are replaced with new ones on upgrade (hardware related ones are not changed).  All the klipper parmeters in mmu_parameters.cfg and macro variables in mmu_macro_vars.cfg (and addon cfg's) are also retained.  If you want to add functionality you should do so in your own cfg file and `[include]` that in your printer.cfg after the other Happy Hare includes. This will keep it separate and will then upgrade cleanly.

Note that nothing is lost .. there will be a date stamped backup of the whole mmu config directory.  Note also that you should be able to include your custom cfg's in that directory if you want to keep functionality together.

closed 2024-12-08 by github-actions[bot]