moggieuk/Happy-Hare #475: Printer automatically resumes after extruder endstop untriggered

wdcocq, 2024-10-21:
I'm not entirely sure if this is a Happy-Hare issue, but I've noticed the following behaviour a few times recently since I'm on the latest Happy-Hare version.

When I notice a filament load failed and I pause the printer before it's done with the user POST_LOAD macro (Just a nozzle scrub). It will pause and move to the front, but it will start resuming by itself once I remove the filament from the toolhead and the extruder endstop goes from triggered to empty. 

After this the printer becomes unresponsive and won't let me pause it, requiring an emergency stop to stop it.

Any pointers what the issue might be? I'm on the latest main branch currently.

wdcocq, 2024-10-21:
After some more testing, this actually happens during mid print too, if I pause and remove the filament. So it's not necessarily linked to the load sequence

moggieuk, 2024-10-25:
Hmmm, I don't have the time at the moment to run through this step-by-step but there are a few strange things:
1) Not sure why the logger is repeating lines. I haven't seen that happen before and it shouldn't do it. Not you problem but would be worth figuring out. How to you restart klipper?  I.e. what button to you hit on Mainsail...?
2) I did recently add a insert / runout trap on the extruder "entry" sensor.  In your log I see this being ingored:
```
14:42:23 > __MMU_EXTRUDER_INSERT
14:42:23   DEBUG: Ignoring extruder insertion because bypass not selected
```
The runout should never been seen accidentally because it is disabled during the load/unload but the INSERT can be and is there to support the bypass functionality.   If you see a "> __MMU_EXTRUDER_RUNOUT" line in your log I'd like to see it (because it will issue an internal klipper resume command.

moggieuk, 2024-10-27:
I'm unsure if it will help, but I just pushed a change that should eliminate "last" runout/insert events.  This might clear up this problem for you. Let me know.

wdcocq, 2024-10-28:
I've tested it and sadly I still get the same results.

moggieuk, 2024-10-28:
On the logger .. I appears you have/had multiple logging threads running.  That would do nothing other than slow down your rpi (and make TTC issues more likely).  I can only occur if klipper does issue a shutdown event but I can't reproduce how you did that!!

The previous message was meant to say any "late" runout/insert events.

Perhaps you can send my another full `mmu.log` -- there might be other clues I can look for

moggieuk, 2024-10-28:
I have an idea of what might be happening...  I think runout/insert events should be ignored during pause.   Normally they are but this is not the case for the newer (extruder sensor) events.  I can address...

wdcocq, 2024-10-28:
It seems you're on the right track, but now klipper crashes :)
This is the exception I got when i tried it:
```
Unhandled exception during run
Traceback (most recent call last):
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/klippy.py", line 176, in run
    self.reactor.run()
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/reactor.py", line 292, in run
    g_next.switch()
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
                            ^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/ls0ai9826hji2w55q6i1ml5fd3c3zy5x-klipper-with-plugins/lib/klipper/extras/mmu_sensors.py", line 72, in _runout_event_handler
    self._exec_gcode(self.runout_gcode + " DO_RUNOUT=1")
                     ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
Transition to shutdown state: Unhandled exception during run
```

wdcocq, 2024-10-28:
> On the logger .. I appears you have/had multiple logging threads running. That would do nothing other than slow down your rpi (and make TTC issues more likely). I can only occur if klipper doesn't issue a shutdown event but I can't reproduce how you did that!!

It's very likely the multiple log entries is something weird about my setup and caused by improper restarts. I was testing/restarting a lot when I fetched those logs. I don't see them anymore now.

moggieuk, 2024-10-28:
I'm an idiot.  Yes I'll fix that code!!!

moggieuk, 2024-10-28:
FIxed. Sorry .. trying to do too many things at once :-)

wdcocq, 2024-10-28:
That seems to have fixed it. Thanks!

closed 2024-10-28 by wdcocq