moggieuk/Happy-Hare #476: Split install.sh script

wdcocq, 2024-10-21:
I've configured my printer with NixOS, including Happy-Hare. But the current implementation of the install script kinda makes this ugly and error prone. So before I spend any time on streamlining this I wonder if there is interest in the following or whether I should keep it on my own fork:

I would propose to make the following changes:
- Split the install scripts in a config questionnaire and install/upgrade part. Where the questionnaire creates a config file for the install/upgrade script to use. This way I can provide the config file in my NixOS configuration and just call the install part.
- I don't want to add any new dependencies, and keep the install steps as close as possible. So I propose to re-use the Kconfig lib that Klipper also uses and provides in it's repo, and use make to call it.  Then the install steps would basically be: 
```sh
make menuconfig   # only on first install
make install      # to install or upgrade
```
- The extra flags -k -c -m, etc can then live in the config file (and be configured through menuconfig) instead of using flags 

So is this something you want to deal with, or prefer to keep it as is? I'm more than willing to write the PR for this!

All suggestions are welcome of course :smile:

wdcocq, 2024-10-25:
I didn't notice there was a dev branch preparing for 3.0. Thanks!
Skimming over the changes, It would probably have broken my setup. So I feel like this change can be worthwhile, and hopefully a bit more maintainable than a big shell script. And from what I got so far, much more configurable if necessary.

>  Be warned there is a A LOT of testing. 

I've noticed, but I feel like setting things up this way makes testing easier,, as you can create different testing configs.
How do you currently test for the different setups? Can't imagine you have every possible combination :)

>  I still have some improvements to make

Anything I can pick up along the way? Or are they only related to Happy Hare functionality

moggieuk, 2024-10-27:
The work on v3 is slowing done now. Just about code complete.  Perhaps this is an idea you can work on in the near future .. I'm definitely prepared to work with it but personally, I need some time back to actually do some printing again rather than writing code.  I want to get v3 out, debugged, then build my next printer :-)

wdcocq, 2024-10-30:
I'm going to close this issue in lieu of the PR #492

closed 2024-10-30 by wdcocq