moggieuk/Happy-Hare #481: MMU_CALIBRATE_TOOLHEAD reports entry_to_extruder = 0

dominikacri, 2024-10-25:
Hi,
I tried MMU_CALIBRATE_TOOLHEAD with ercf2 and an toolhead entry sensor.

extruder_to_nozzle and sensor_to_nozzle seems valid but entry_to_extruder is 0 and should be around 25. 

Calibration Results (clean nozzle):
> toolhead_extruder_to_nozzle: 93.8 (currently: 94.5)
> toolhead_sensor_to_nozzle: 65.5 (currently: 66.7)
> toolhead_entry_to_extruder: 0.0 (currently: 25.0)

[mmu.log](https://github.com/user-attachments/files/17524921/mmu.log)
[mmu_config.zip](https://github.com/user-attachments/files/17524945/mmu_config.zip)

moggieuk, 2024-10-27:
The first problem is see is the extruder sensor is triggering A LOT.  Are you sure this is set up correctly? 
```
17:51:30 > __MMU_EXTRUDER_RUNOUT
17:51:32 > __MMU_EXTRUDER_RUNOUT
17:51:45 > __MMU_EXTRUDER_RUNOUT
17:51:46 > __MMU_EXTRUDER_RUNOUT
17:51:47 > __MMU_EXTRUDER_RUNOUT
17:51:48 > __MMU_EXTRUDER_RUNOUT
17:51:52 > __MMU_EXTRUDER_RUNOUT
17:51:54 > __MMU_EXTRUDER_RUNOUT
17:51:57 > __MMU_EXTRUDER_RUNOUT
17:51:58 > __MMU_EXTRUDER_RUNOUT
17:51:59 > __MMU_EXTRUDER_RUNOUT
17:52:18 > __MMU_EXTRUDER_RUNOUT
17:52:22 > __MMU_EXTRUDER_RUNOUT
17:56:38 > __MMU_EXTRUDER_INSERT
17:56:38   DEBUG: Ignoring extruder insertion because bypass not selected
```
Maybe your were inserting and removing filament and it is expected but I thought I'd check.   If this sensor isn't working reliably then the calibration cannot work.

dominikacri, 2024-10-28:
Not in that small timeframe. Think you are right I will check my sensor. Thanks a lot !

closed 2024-10-28 by dominikacri