moggieuk/Happy-Hare #482: Printer keeps pausing for no reason

SHKinsem, 2024-10-26:
Here is the extracted info from mmu.log
```
> MMU_SLICER_TOOL_MAP SPARSE_PURGE_MAP=1 NUM_SLICER_TOOLS=6
18:46:02 --------- Slicer MMU Tool Summary ---------
         3 color print (Purge volume map loaded)
         T0 (Gate 0, PETG, d50000, 250°C)
         T3 (Gate 3, PETG, 101010, 250°C)
         T4 (Gate 4, PETG, ffffff, 248°C)
         Initial Tool: T3
         -------------------------------------------
         Purge Volume Map (mm^3):
         To -> T0   T1   T2   T3   T4   T5 
         T0    -    -    -   188  685   - 
         T1    -    -    -    -    -    - 
         T2    -    -    -    -    -    - 
         T3   423   -    -    -   642   - 
         T4   335   -    -   181   -    - 
         T5    -    -    -    -    -    - 
18:49:33 Loading initial tool T3...
18:49:33 > MMU_CHANGE_TOOL STANDALONE=1 TOOL=3
18:49:33   DEBUG: Tool change initiated with standalone MMU tip forming
18:49:33 Tool T3 is already loaded
18:49:33 > MMU_CHANGE_TOOL TOOL=3 NEXT_POS="130.218,73.027" ; T3
18:49:33   DEBUG: Tool change initiated with standalone MMU tip forming
18:49:33 Tool T3 is already loaded
18:50:10 > PAUSE
18:50:10   DEBUG: MMU PAUSE wrapper called
18:50:10   DEBUG: Saving toolhead gcode state and position (X:132.8 Y:75.0 Z:0.2 E:0.2) for pause
18:50:10 Saving toolhead position (x:132.8, y:75.0, z:0.2)
18:50:10 Retracting 2.0mm
18:50:10 Parking toolhead at (x:50.0, y:50.0, z:5.2) for pause operation
18:50:29 > RESUME
18:50:29   DEBUG: MMU RESUME wrapper called
18:50:29   DEBUG: Continuing from printing state after resume
18:50:29 Restoring toolhead position to (last) (x:132.8, y:75.0, z:0.2)
18:50:29 Un-retracting 2.0mm
18:50:29   DEBUG: Ensuring correct gcode state and position (X:132.8 Y:75.0 Z:0.2 E:84.7) after resume
18:50:32 > PAUSE
18:50:32   DEBUG: MMU PAUSE wrapper called
18:50:32   DEBUG: Saving toolhead gcode state and position (X:159.8 Y:70.5 Z:0.0 E:10.6) for pause
18:50:32 Saving toolhead position (x:159.8, y:70.5, z:0.0)
18:50:32 Retracting 2.0mm
18:50:32 Parking toolhead at (x:50.0, y:50.0, z:5.0) for pause operation
18:50:36 > RESUME
18:50:36   DEBUG: MMU RESUME wrapper called
18:50:36   DEBUG: Continuing from printing state after resume
18:50:36 Restoring toolhead position to (last) (x:159.8, y:70.5, z:0.0)
18:50:36 Un-retracting 2.0mm
18:50:36   DEBUG: Ensuring correct gcode state and position (X:159.8 Y:70.5 Z:0.0 E:95.1) after resume
18:50:43 > PAUSE
18:50:43   DEBUG: MMU PAUSE wrapper called
18:50:43   DEBUG: Saving toolhead gcode state and position (X:158.8 Y:69.3 Z:0.0 E:40.1) for pause
18:50:43 Saving toolhead position (x:158.8, y:69.3, z:0.0)
18:50:43 Retracting 2.0mm
18:50:43 Parking toolhead at (x:50.0, y:50.0, z:5.0) for pause operation
```

I also checked the klippy.log

```
Pausing toolhead to calculate extruder phase offset
Filament Sensor toolhead_sensor: runout event detected, Time 4161.41
```

As you can see in the time stamp, it just keeps pausing for no reason. No pop-out message, neither in the console.
This happens after updating to 2.7.2.

And I have to complain a bit. Every single time I update the happy hare, it will spend me hours to reconfigure something, weird bugs keep popping out until I carefully checked the doc. This includes the changes in client macros, and the start/end gcodes in slicer, etc. Please think twice when introducing some new features and take the backward compatibility into serious consideration. 😢

SHKinsem, 2024-10-26:
Tested a few more config combinations.
Disabling MMU did not solve the problem but switch off the tool head sensor did. No idea what is going on...

moggieuk, 2024-10-27:
You know, I'd appreciate the absence of "complaints".  This problem is a project of love. I've put many hundreds of hours into it with little reward and as a single person operation I can't possibly (or want to) test all possible setups.  You obviously have not looked to see how much work is put into supporting upgrades -- I try really hard whilst still wanting to continue the evolution. It cannot be perfect and that my friend is the nature of open source.

Now as for your problem.  Happy Hare never calls "PAUSE" directly.  My best guess is that you have a standalone toolhead sensor or extruder_sensor configuration in your `printer.cfg`.  That configuration is calling pause.  Can you confirm that you have extruder sensors defined in the `[mmu_sensors]` section of `mmu_hardware.cfg`, like:
```
[mmu_sensors]
extruder_switch_pin: {extruder_sensor_pin}
toolhead_switch_pin: {toolhead_sensor_pin}
```

o1hitman1o, 2024-10-27:
I agree more info is needed. What is the current setup that you have?  Is your slicer calling the pause?

SHKinsem, 2024-10-27:
Thanks for the reply. We would appreciate your work into this. I have been using your software since the ercf3.0 and seen what you have achieved. I did not mean to complain all the work you have done but just want to spot some problems when this project is going. 

As for the problem, I know where it come from now thanks to your reminder. Every time the happy hare updates, it will skip the mmu_hardware.cfg. I just checked that file, and it turned out to be a very old config with the filament sensor:

```
[filament_switch_sensor toolhead_sensor]
switch_pin: ^!TOOLHEAD_SENSOR
```

Now my printer is still printing something, after that finished I will give it a try.

SHKinsem, 2024-10-27:
BTW I want to address another problem I encountered during this update. Looks like the toolchange parking feature is turned on by default and I did not notice that. Now it results in a ton of stringy in the print. 

Personally, I am a studying in a design related major and I understand people hate us designers lol. But just a warm reminder, do consider the user experience and minimize the impact of every update, klipper does it quite well though it also ignored/closed a lot of interesting PRs. And sorry for my bad temper in the beginning since at that time I also spent like 5 hours to debug it yet had no idea what was going on.

SHKinsem, 2024-10-27:
Problem solved after defining the [mmu_sensors].
And it looks like I need to update the mmu.cfg and mmu_hardware.cfg thoroughly myself.
I am closing this issue.

closed 2024-10-27 by SHKinsem