moggieuk/Happy-Hare #483: Bugs in development branch

3DCoded, 2024-10-26:
Thank you for writing this amazing MMU driver!

I am trying to use a Type-B setup (the [3MS](https://github.com/3DCoded/3MS)) in the `development` branch, and came across a few main issues that are preventing me from moving forward. 

The first issue I encountered is that you can't use gear steppers on multiple MCU's. My setup uses two gear steppers on one MCU (Einsy RAMBo) and the third on the MMU's MCU (SKR Mini E3 V2). When I try to configure these motors, it gives me this error:
```
Multi-mcu homing not supported on multi-mcu shared axis
```

The next error I encountered is that you can't use multiple TMC driver types (T0 and T1 use TMC2130s, and T2 uses a TMC2209). The error I get when I try this is:
```
No section `tmc2130 stepper_mmu_gear_2`
```

The last problem I encountered is that I am unable to run `MMU_LOAD` or `MMU_UNLOAD` with only one filament sensor configured (an extruder entry sensor). The error I get is:
```
MMU issue: Load sequence failed: Attempting to load gate but gate sensor 'mmu_post_gate_0' is not configured on MMU!.
Occured when loading tool: > T0
```

Any help would be appreciated!

3DCoded, 2024-10-26:
I was able to (sort of) bypass the last issue with the filament sensors by defining a virtual pin as the post gate sensor (and disabling T2). Now I'm getting an internal error on `MMU_EJECT`.

[mmu.log](https://github.com/user-attachments/files/17530938/mmu.log)
[klippy (3).log](https://github.com/user-attachments/files/17530937/klippy.3.log)

3DCoded, 2024-10-26:
I was able to (I think) find the cause of the second error. In `mmu_machine.py`, lines `122-124`, this:

```
base_tmc = '%s %s' % (chip, GEAR_STEPPER_CONFIG)
if config.has_section(base_tmc):
    tmc_section = '%s %s_%d' % (chip, GEAR_STEPPER_CONFIG, i)
```

can be changed to this:

```
base_tmc = '%s %s' % (chip, GEAR_STEPPER_CONFIG)
tmc_section = '%s %s_%d' % (chip, GEAR_STEPPER_CONFIG, i)
if config.has_section(base_tmc) and config.has_section(tmc_section):                   
```

moggieuk, 2024-10-27:
Hi! Firstly the "development" branch is often unstable. I'm getting close to code complete, but not there yet.

That said, since you found it, perhaps you can send a private message of Discord.  Once connected I can create a small group of folks testing with "Type-B" designs.  I think that would be the most efficient...

chadakenn, 2024-10-27:
I'm also working on a 3ms and am willing to test so add me in the discord 

> Hi! Firstly the "development" branch is often unstable and not "public" yet. I'm getting close to code complete, but not there yet.
> 
> That said, since you found it, perhaps you can send a private message of Discord. Once connected I can create a small group of folks testing with "Type-B" designs. I think that would be the most efficient...

moggieuk, 2024-10-28:
What is your discord name?  Perhaps you can drop me a DM.  @moggieuk on discord.

chadakenn, 2024-10-28:
> What is your discord name? Perhaps you can drop me a DM. @moggieuk on discord.

Sent

3DCoded, 2024-10-28:
I sent the DM.

moggieuk, 2024-10-28:
Ok. Closing this issue, since v3 hasn't been released yet :-)

closed 2024-10-28 by moggieuk