moggieuk/Happy-Hare #484: [request] Single Color Prints Should Not Unload Filament When Already Loaded

Imperite, 2024-10-27:
Currently still doing a lot of repeated, single-color prints until I can get more time to tune my ERCF to work reliably. When I go to start a single color print, since I leave the filament loaded, it automatically ejects the current tool, checks that tool, [does print_start.] and then re-loads the same tool. 
I know it's not a massive issue, but would it be feasible to add a check for using the mmu as a passthrough during print start? I.e. when it detects it's only one tool and that tool is already loaded, just skip the unload/check gate sequence. I love the other features (runout/clog detection) so don't want to fully disable yet. Still also getting unloads/loads fully consistent, which is why I don't have it auto-unload on print end.

I *think* it could be as simple as adding a condition to the `cmd_MMU_CHECK_GATE` in `~/Happy-Hare/extras/mmu.py` or modifying `MMU_START_CHECK` in `mmu/base/mmu_software.cfg` to compare loaded tool and needed tools, but am not sure if there would be any other implications of that - especially on mmu_check_gate.

### Expected Behavior:
On print starts, mmu detects single-color print with filament already loaded. Does not unload filament and proceeds directly print.

On  Happy Hare `v2.7.2-31-gab85033a`; included my mmu.log (ignore the issues earlier on lol - that's stuff I still have to tune. Last commands starting around 18:03:52 is where the issue is most clearly documented)
[mmu.log](https://github.com/user-attachments/files/17535715/mmu.log)

Is this a reasonable request? Not sure how much it would complicate things, but would be nice QoL feature.

moggieuk, 2024-10-28:
There is a "bypass" gate for this very purpose.  Most folks don't generally leave the filament loaded when in the MMU but if you do you can disable the "gate check"  in this section of `mmu_macro_vars.cfg`:
```
# These variables control the behavor of the MMU_START_SETUP and MMU_START_LOAD_INITIAL_TOOL macros
variable_user_pre_initialize_extension      : ''        ; Executed at start of MMU_START_SETUP. Commonly G28 to home
variable_home_mmu                           : False     ; True/False, Whether to home mmu before print starts
variable_check_gates                        : True      ; True/False, Whether to check filament is loaded in all gates used
variable_load_initial_tool                  : True      ; True/False, Whether to automatically load initial tool
```

You should also be able to disable the MMU with `MMU ENABLE=0` for it to ignore normal behavor.  I have to admit I haven't tested this latter option but I think it would work.

Imperite, 2024-10-29:
Thank you, forgot about the bypass. I set up my mmu before the new wiki so didnt see the page on it then. Will use that until I can finish tuning it and bring it back to more 'normal' operation, then!

closed 2024-10-29 by Imperite