moggieuk/Happy-Hare #485: Issue with update 2.7.3

Marck80, 2024-10-28:
Thank you for this fantastic work.
I am writing to report a bug that I encountered while attempting to update. Specifically, I receive an error on line 798 during the update process.

```
Running on 'main' branch
Checking for updates...
Found a new version of Happy Hare on github, updating...
Now on git version v2.7.3-13-g0bf0e5d
Running the new install script...
Klipper config directory (/home/biqu/printer_data/config) found
Klipper klipper.service systemd service found
Reading default configuration parameters...
Reading mmu_parameters.cfg configuration from previous installation...
Reading mmu_software.cfg configuration from previous installation...
Reading mmu_sequence.cfg configuration from previous installation...
Reading mmu_cut_tip.cfg configuration from previous installation...
Reading mmu_form_tip.cfg configuration from previous installation...
Reading mmu_macro_vars.cfg configuration from previous installation...
Reading blobifier.cfg configuration from previous installation...
Reading mmu_erec_cutter.cfg configuration from previous installation...
Setting Blobifier variable_z_raise and variable_purge_length_maximum from previous settings
/home/biqu/klipper/Happy-Hare/install.sh: line 798: python: command not found
biqu@BTT-CB1:~/klipper/Happy-Hare$ nano /home/biqu/klipper/Happy-Hare/install.sh

```

Marck80, 2024-10-28:
triangular is defined later
```
    # Blobifer update - Oct 13th 20204
    if [ ! "${variable_iteration_z_raise}" == "" ]; then
        echo -e "${INFO}Setting Blobifier variable_z_raise and variable_purge_length_maximum from previous settings"
        variable_z_raise=$(python -c "print(${variable_iteration_z_raise} * ${variable_max_iterations_per_blob} - $(triangular $((variable_max_iterations_per_blob - 1))) * ${variable_iteration_z_change})")
        variable_purge_length_maximum=$(python -c "print(${variable_max_iteration_length} * ${variable_max_iterations_per_blob})")
    fi
}

triangular() {
    awk "BEGIN {print ($1 * ($1 + 1)) / 2}"
}
```

moggieuk, 2024-10-28:
Hmmm.  Looks like python is not in your PATH.     You can edit and change "python" to "/usr/bin/python" or if that fails "/usr/bin/python3".

I'll work with @Dendrowen and try to avoid using python and update...

wdcocq, 2024-10-28:
Changing it to:
```bash
	if [ ! "${variable_iteration_z_raise}" == "" ]; then
		log_info "Setting Blobifier variable_z_raise and variable_purge_length_maximum from previous settings"
		variable_z_raise=$((variable_iteration_z_raise * variable_max_iterations_per_blob - $(triangular $((variable_max_iterations_per_blob - 1))) * variable_iteration_z_change))
		variable_purge_length_maximum=$((variable_max_iteration_length * variable_max_iterations_per_blob))
	fi

```
Should work, no need for python to do calculations
I was just parsing/changing this code anyway for my upcoming PR :)

moggieuk, 2024-10-28:
I just pushed an update that uses "awk" instead of "python" to avoid PATH issues.  Basically it is one time logic anyway.

moggieuk, 2024-10-28:
Unfortunately floating point calcs are difficult in borne shell.

Marck80, 2024-10-28:
I had /usr/bin/python3 and not python… I created a symbolic link (ln -s /usr/bin/python3 /usr/bin/python) and it worked.

closed 2024-12-11 by github-actions[bot]