moggieuk/Happy-Hare #486: HH 2.7.3 z-hopping when park_toolchange is set to -1,-1 to disable move / park

ningpj, 2024-10-29:
Use case - standard, purge block based filament changes, no blobifier or cutter.

Setting variable_park_toolchange        : -1,-1, 0, 0, 2 to prevent HH from moving/parking on toolchange so it remains above the purge block now causes it to z-hop 2mm following 2.7.3 update. Presume variable_min_toolchange_z       : 2.0 now enforces this when z <= min height whether parking or not. Since I'm not moving the head from the purge block min_toolchange logic should be conditional and only applied when x & y >= 0 or when z-hop is explicitly set.

14:42:05 // Parking toolhead at (x:n/a, y:n/a, z:2.0) for toolchange operation
14:42:05 // Restoring MMU gear stepper run current to 100% configured
14:42:08 // Run Current: 0.89A Hold Current: 0.21A
14:42:08 // Unsynced MMU from extruder
14:42:08 // Unloading filament...

moggieuk, 2024-10-30:
You are correct that it would be the min_toolchange_z "floor" that would cause this min height as a safety feature.

So are you suggesting that if park x,y is -1, -1 AND z-hop is 0 then the min_floor should not be applied?

ningpj, 2024-10-30:
Yes.  z-hopping over the purge block amplifies blobs and would return this to the previous behaviour.  If the toolhead doesn't need to move and all 3 arguments are set to off then the min_floor shouldnt apply

moggieuk, 2024-10-31:
This thinking makes sense but it does potentially defeat the purpose of the floor (which was btw a user suggestion after they scratched their bed).  However if coupled with no X/Y movement I think it is fine.

moggieuk, 2024-10-31:
Since I'm buried in v3.0 development and testing, can you try this fix for me. I think it is as easy as this.  Edit `mmu_sequence.cfg` and add the three lines indicated.  This is in the `[gcode_macro _MMU_PARK]` macro definition around line 125:
```
    {% set park_z_hop = park_z_hop|abs %}
    {% set z_hop_ramp = z_hop_ramp|abs %}
    {% set min_toolchange_z = vars.min_toolchange_z|default(1)|float|abs %} 
    {% if x == -1 and y == -1 and  park_z_hop == 0 %} # ADD ME
        {% set min_toolchange_z = 0 %}                # ADD ME
    {% endif %}                                       # ADD ME
    {% set park_travel_speed = vars.park_travel_speed|default(200)|float * 60 %}
    {% set lift_speed = vars.lift_speed|default(15)|float * 60 %}
    {% set pos = printer.gcode_move.gcode_position %}
    {% set origin = printer.gcode_move.homing_origin %}
    {% set max = printer.toolhead.axis_maximum %}
 ```
 If you verify I will make this change.

ningpj, 2024-12-02:
rolled up into HH 3.0.1 release. tested and confirmed

closed 2024-12-02 by ningpj