moggieuk/Happy-Hare #487: Mainsail and Fluidd history

Spydyr, 2024-10-30:
So I believe I found an issue with the history. 

If you have 
[file_manager]
enable_object_processing: False

And

[mmu_server]
enable_file_preprocessor: True

Then print history works as normal. However larger gcode files may not have the thumbnail. That's not the issue. If both are set to True history records the gcode prints as not the here including the one the has is currently printing. With both set to True though thumbnails always work. This problem started with 2.7.2 as I back tracked to when it was working fine. I have cleared history and gcode files to see if it was "overloaded" and that didn't correct it.

ningpj, 2024-10-30:
Missing thumbnails usually means the pre processor has failed to parse or has timed out parsing the gcode

Do you have ```
[file_manager]
default_metadata_parser_timeout: 30```

in moonraker.conf to increase the default timeout from 10sec?

Spydyr, 2024-10-30:
No I do not but that is a minor issue. The big issue is history not showing the files are missing even the one that is printing with both set to True. Could that timeout be part of it?

Spydyr, 2024-10-30:
https://drive.google.com/file/d/1MtbN-Zlt5R4S6PCnAZKAnIK56_xqZsX5/view?usp=drivesdk

Here I the pic I took to show. The top one was printing when I took the photo.

ningpj, 2024-10-30:
Highly likely as the preprocessor essentially rewrites the gcode file to expand HH tokens like ``!referenced_tools!`` ``!colors!``, ``!temperatures!``, etc.  If it times out, the resulting gcode will be corrupt/incomplete.

Recommend you quickly override the default moonraker timeout to 30sec and retest - upload a new, sizable gcode file you have been having issues with and confirm the thumbnail is visible (might take > 10sec depending on the size of the file and SD card performance).  Export the parsed gcode and check the HH tokens have also been expanded and replaced with expected values.

ningpj, 2024-10-30:
If you still have issues after setting this, please upload an extract from your moonraker.log and confirm your moonraker version.  this is what you are looking for - Running MMU enhanced .... block 
```
2024-10-30 18:15:46,855 [application.py:post()] - Processing Uploaded File: MM_Test_print_9_colors_ABS_0.4_1h12m.gcode
2024-10-30 18:15:47,191 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2024-10-30 18:15:47,198 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

2024-10-30 18:15:47,215 [shell_command.py:pipe_data_received()] - INFO:metadata:File 'MM_Test_print_9_colors_ABS_0.4_1h12m.gcode' currently supports cancellation, processing aborted

2024-10-30 18:15:47,302 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/pi/printer_data/gcodes/MM_Test_print_9_colors_ABS_0.4_1h12m.gcode

2024-10-30 18:15:47,624 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 0.32s. Detected gcode by slicer: OrcaSlicer

2024-10-30 18:15:47,868 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Writing MMU placeholders,Inserting next position to tool changes took 0.24s

2024-10-30 18:15:47,921 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/mmu_server.py -m -n -p /home/pi/printer_data/gcodes -f "MM_Test_print_9_colors_ABS_0.4_1h12m.gcode" --check-objects) successfully finished
2024-10-30 18:15:47,927 [application.py:log_request()] - 201 POST /api/files/local (192.168.2.185) [_TRUSTED_USER_] 1193.44ms
```

Also a sliced and unprocessed gcode file that has this issue so we can verify what your are experiencing (slice and export directly to a file)

Spydyr, 2024-10-30:
I will try it out after the current print finishes in about 8 hours.

Spydyr, 2024-10-30:
[moonraker (3).log](https://github.com/user-attachments/files/17574099/moonraker.3.log)

This is the latest log 4 attempts with the same file that shows the thumbnail regardless. History show the file does not exist when printing the file. And now it doesn't matter whether enable_object_processing is True or False.

ningpj, 2024-10-30:
OK, it looks like you have spoolman integration enabled which maybe the issue here given recent updates to the preprocessor to support this.  Can you upload the unprocessed ``B_1_Yellow_PLA_5h19m.gcode`` file (export from slicer)?

Spydyr, 2024-10-30:
https://drive.google.com/file/d/1U8eLYl4KQYVXj6BDns9_oJv5N6jtyqNc/view?usp=drive_link

Here is the link to it. It is like 25mb.

Spydyr, 2024-10-30:
Now if I rescan metadata after the upload then history works as it should. Also I've only been running spoolman for 2 weeks and this issue by timeline started well before that.

ningpj, 2024-10-30:
I rarely use history and had a look on my other printer (v0.2) which doesn't have HH installed and can see same issue with missing icons for some jobs although others are fine (all sliced with the same 2.1.1 version of orcaslicer).   Maybe a moonraker issue or orca related and a problem with the small embedded icon. Unsure what the cause is at this stage.  

Your gcode file parsed fine on my HH system with tokens correctly expanded and icon visible in the jobs queue.  However after submitting and cancelling the job its icon isnt visible in the History queue.  Snap from my v0.2 showing the same issue in fluidd:
[image]

Spydyr, 2024-10-30:
I use the history for seeing my most popular prints and for tweaking slicers so the times are close.  And keep track of actual Total Print Time. I tried in Fluidd too and like you I got the same thing. I'm just not sure if it is a moonraker issue or a HH. I have thought about downgrading back to 2.7.1 because it was working with it.

ningpj, 2024-10-30:
As mentioned i see this issue on my v0.2 which doesnt have HH installed so definitely not caused by HH

Spydyr, 2024-10-30:
My Sapphire doesn't have the issue and it doesn't have hh this is why I'm confused. But I do think it's more moonraker metadata issue than anything. There is something different and I just can't find it

moggieuk, 2024-11-03:
When HH will do a metadata scan whenever the file is created or changed.  It piggybacks on the very same metadata scan that moonraker would have ran anyway.  However if parsing does result in modification of the file then it would appear as "new" (not in print history).  Not sure what might be happening without research but I can't do that ATM because I'm deep into Happy Hare v3.0.  I'll keep this issue open to track.  In the mean time if you or @ningpj  have any more insights I'm all ears...

Spydyr, 2024-11-03:
I just checked and no dates are not changing after printing. Also my Sapphire does not have spoolman installed on it. I just finished a print with it. I will test installing spoolman and loading a print. I will also be install long HH on it as I have to remove my ERCF from my Voron because I will be printing PPS-CF for the next week and can run it through there. That will give another test point. If it is strictly related to my Voron I will wipe it completely out and start over. 

I will keep you informed of my findings over the next week.

SHKinsem, 2025-02-23:
Same problem, fresh update from 2.7.3. Partial moonraker log is attached

```
2025-02-23 19:15:45,416 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-czrw62io.json) exited with return code 2
2025-02-23 19:15:45,418 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-02-23 19:15:45,881 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2025-02-23 19:15:45,894 [shell_command.py:pipe_data_received()] - usage: mmu_server.py [-h] [-f <filename>] [-p <path>] [-u <ufp file>] [-o]
                     [-m] [-n]

2025-02-23 19:15:45,895 [shell_command.py:pipe_data_received()] - mmu_server.py: error: unrecognized arguments: -c /tmp/metacfg-1t7584so.json

2025-02-23 19:15:46,015 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-1t7584so.json) exited with return code 2
2025-02-23 19:15:46,017 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-02-23 19:15:46,504 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2025-02-23 19:15:46,517 [shell_command.py:pipe_data_received()] - usage: mmu_server.py [-h] [-f <filename>] [-p <path>] [-u <ufp file>] [-o]
                     [-m] [-n]

2025-02-23 19:15:46,518 [shell_command.py:pipe_data_received()] - mmu_server.py: error: unrecognized arguments: -c /tmp/metacfg-lslzkj3b.json

2025-02-23 19:15:46,637 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -c /tmp/metacfg-lslzkj3b.json) exited with return code 2
2025-02-23 19:15:46,638 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2566, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2621, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.exceptions.ServerError: Extract Metadata returned with error
2025-02-23 19:15:46,639 [file_manager.py:_process_metadata_update()] - Unable to extract medatadata from file: OrcaCube_PETG_26m52s.gcode
2025-02-23 19:15:46,646 [application.py:log_request()] - 201 POST /api/files/local (192.168.1.2) [_TRUSTED_USER_] 2235.58ms
```

Don't know where did that extra argument '-c' come out

Spydyr, 2025-03-08:
I gave up after 5 complete reinstalls and going through them one by one and I just get hit and misses with the metadata. lately it has been all misses.

Spydyr, 2025-03-08:
I will text this later as I have a multi part multi color print I'm starting in a few hours.

Spydyr, 2025-03-09:
After some testing it seems to have fixed the issue. Going to do an 8 color print later this week and let you know the results.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

You confirmed updating Happy Hare fixed this, and v4 also handles the newer way Moonraker calls the metadata parser.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk