moggieuk/Happy-Hare #493: Calibration settings file (mmu_vars.cfg) not found.

MickaelVanSchie, 2024-10-31:
I keep running into the issue that Happy-Hare doesn't find thie file mmu_vars.cfg.

Klipper returns this error:
`Calibration settings file (mmu_vars.cfg) not found. Check [save_variables] section in mmu_macro_vars.cfg
Also ensure you only have a single [save_variables] section defined in your printer config and it contains the line: mmu__revision = 0. If not, add this line and restart

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted`

If I look in `mmu_macro_vars.cfg` I have this lines:

`[save_variables]
filename: ~/printer_data/config/mmu/mmu_vars.cfg`
 
Switching the filename to the absolute path does not change anything. 

The contents of ~/printer_data/config/mmu/mmu_vars.cfg is:

`
# This is the template file for storing Happy Hare state and calibration variables. It is pointed to
# with the [save_variables] block in 'mmu_macro_vars.cfg'
#
# If you want to use an existing "variables" file, then that is fine but make sure you copy the
# "mmu__revision" line to it because Happy Hare will look for this to validate correct setup
#
[Variables]
mmu__revision = 0
`
And a `ls -la` within the mmu folder returns this:

drwxr-xr-x 5 pi pi 4096 Oct 31 11:35 .
drwxr-xr-x 7 pi pi 4096 Oct 31 11:32 ..
drwxr-xr-x 2 pi pi 4096 Oct 31 09:55 addons
drwxr-xr-x 2 pi pi 4096 Oct 31 11:34 base
-rw-r--r-- 1 pi pi  382 Oct 31 11:35 mmu_vars.cfg
drwxr-xr-x 2 pi pi 4096 Oct 31 10:16 optional

My klippy.log ends with:

`
MMU Log: /home/pi/printer_data/logs/mmu.log
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 135, in _connect
    cb()
  File "", line 970, in handle_connect
    raise self.config.error("Calibration settings file (mmu_vars.cfg) not found. Check [save_variables] section in mmu_macro_vars.cfg\nAlso ensure you only have a single [save_variables] section defined in your printer config and it contains the line: mmu__revision = 0. If not, add this line and restart")
configparser.Error: Calibration settings file (mmu_vars.cfg) not found. Check [save_variables] section in mmu_macro_vars.cfg
Also ensure you only have a single [save_variables] section defined in your printer config and it contains the line: mmu__revision = 0. If not, add this line and restart
`

moggieuk, 2024-11-02:
Is it possible you have another `[save_variables]` definition elsewhere in your `printer.cfg`?  Klipper will only allow one, so if you have another you need to merge them into one.

Maybe try changing:
```
[save_variables]
filename: ~/printer_data/config/mmu/mmu_vars.cfg
```
to an explicit absolute path (to remove different user accounts for being a possible source of error)
```
[save_variables]
filename: /home/pi/printer_data/config/mmu/mmu_vars.cfg
```

Edit. I just read you tried absolute path so I have to believe klipper is looking at a different variables file.

MickaelVanSchie, 2024-11-02:
@moggieuk I feel stupid. I'm running RatOS, and I've been looking a dozen times through all files not being able to find it.. But I just did. Ticket can be closed.

RustysGarage, 2024-11-10:
i have this same issue with ratos 2.1 not sure how to fix it

MickaelVanSchie, 2024-11-11:
@RustysGarage 

RatOS uses a [Variables] definition itselve. The variables for RatOS are stored in the root directory in a file called ratos-variables.cfg.

This means that you can remove the definition of the variables file from mmu_macro_vars.cfg

You will also need to add `mmu__revision = 0` to the defined [Variables] blockin ratos-variables.cfg

closed 2024-11-11 by MickaelVanSchie