moggieuk/Happy-Hare #498: Park_pos ??

EdwinPock, 2024-11-03:
"Warning: Park_pos (58.0mm) cannot be greater than 'toolhead_extruder_to_nozzle' distance of 41.0mm! Assumming fully unloaded from extruder"

Can someone tell me what Park_Pos is going on and where can I find?

moggieuk, 2024-11-04:
Are tip-forming or tip-cutting?  

If cutting this variable is dynamically set by the macro.  It suggests that either 'variable_rip_length' or 'variable_blade_pos` are incorrect.

If tip-forming then this will calculated based on actual extruder movement.  This error suggests that your tip forming variables are causing the filament to end up outside of the extruder.  I.e. the park pos is the anticipated end of filament in the extruder measured from nozzle.  If greater than `toolhead_extruder_to_nozzle` then it is outside of extruder. Also check that `toolhead_extruder_to_nozzle` is correct -- it seems a little short.

EdwinPock, 2024-11-04:
Variable_blade_pos: 48. This is exactly the value and is short, because it is an extruder Matrix from Trianglelab, which I transformed. In which I installed a cutting knife. 
variable_rip_length: 10 When the variable was 2, he also reported an error.
Although I write the truth, the rabbit does not like something. :-)

EdwinPock, 2024-11-08:
Well, I'll explain it. Due to the design of my extruder, you cannot install the knife below extruder_gears. I can install the knife only above the knurl. Therefore, in my case, "park_pos" will always be greater than "toolhead_extruder_to_nozzle". I understand that my assembly method was not considered. But can this error be turned off or excluded somehow?

moggieuk, 2024-11-12:
Oh, I see.  The current way unloading works can't support this position for cutting.  Even if it could you would end up with a lose piece of filament above the extruder gears that is likely to create all kinds of problems.

Maybe you could experiement by creating a minimal tip forming macro (very simple) and call that say MY_FORM_TIP.  Then set:
```
form_tip_macro: MY_FORM_TIP
```
Then you have to turn on the "macro unload sequence" and code an extra step after `_MMU_STEP_UNLOAD_EXTRUDER` to move filament back into the cutter and cut

EdwinPock, 2024-11-12:
In my case, it works very stable, the distance from the knife to the extruder's gear is only 8 mm and it is best that after cutting it does nothing from the filament and try to get such an effect, or that after cutting the filament is well anchored in the extruding gear. In addition, when charging, I have synchronization and powerful engines on, which helps to load the filament significantly. 
The only thing that irritated me were these warnings. :-)

moggieuk, 2024-11-12:
@EdwinPock I had another thought.. don't know why this didn't occur to me earlier.  While I still believe cutting outside of the extruder is a bad thing, you can make it work....

So make a copy of `_MMU_CUT_TIP` macro.  Rename it to say `MY_CUT_TIP`, point HH to use this as described before.

Then in your copy, edit this line:
```
    # Dynamically set the required output variables for Happy Hare
    SET_GCODE_VARIABLE MACRO=_MMU_CUT_TIP VARIABLE=output_park_pos VALUE={extruder_park_pos}
```

This is how the macro reports back the position to HH.  You can thus hard code this to report a value to keep HH happy ;-)

PocketMiner82, 2025-02-17:
Hey, I've had the same problem and just opened a PR (#640) that adds support for cutters placed before the extruder. It is currently working very well but not tested enough to be ready to be merged (I think).

closed 2026-10-21 by moggieuk