moggieuk/Happy-Hare #499: Where is the carrot?

EdwinPock, 2024-11-03:
After the last update, the carrot icon disappeared on the display, as I press it, I enter the main panel but the carrot disappeared. 
How to recover it?

ningpj, 2024-11-03:
Rerun the installer as a few icon's have changed to make HH more generic and less ERCF specific....The icon is now a pink happy hare :-)
```
cd ~/KlipperScreen/happy_hare
./install_ks.sh -g {num gates}
```

moggieuk, 2024-11-04:
Yeah, as much as I like the carrot (and has a place in my heart) I will be releasing Happy Hare v3.0 soon which will extend support to many other MMU designs.

EdwinPock, 2024-11-04:
Thanks it worked :

The Matrix: "Follow the Pink Rabbit" :-)

closed 2024-11-05 by EdwinPock

cin3m, 2024-11-09:
Just an idea that came up my mind (I don't know how complicated the implementation could be): Can you enable the user to choose the desired icon as an optional step within the wizard?

Kind Regards

Whitsend1000, 2024-11-09:
iS THIS THE PROPER CODE FOR 8 GATES: 
cd ~/KlipperScreen/happy_hare
./install_ks.sh -g {8}