moggieuk/Happy-Hare #511: sync_multiplier_low error HH v2.7.3-25

Ekodas, 2024-11-11:
After updating to HH v2.7.3-25-g70115862, I now receive the error:

Option 'sync_multipler_low' is not valid in section 'mmu'

In order to run, I now need to comment out the line in mmu_parameters.cfg like below in order for Klipper to start.

sync_feedback_enable: 1			# 0 = Turn off (even with fitted sensor), 1 = Turn on when printing
sync_multiplier_high: 1.05		# Maximum factor to apply to gear stepper `rotation_distance`
#sync_multipler_low: 0.95		# Minimum factor to apply

Tannoo, 2024-11-11:
Ditto of above.

mrmultifunk, 2024-11-11:
Check the spelling, sync_multipler_low --> sync_multiplier_low.

Ekodas, 2024-11-11:
Good catch mrmultifunk!  It appears that HH v2.7.3-25-g70115862 has a type-o in it.  I'm in the middle of a print right now, but I'll fix the type-o and see if that fixes the issue appropriately.

moggieuk, 2024-11-12:
Yes it was an typo.  It is fixed now with the correct spelling.  Sorry about that.

closed 2024-12-15 by moggieuk