moggieuk/Happy-Hare #512: The use of demultiplexers.

porar-oss, 2024-11-12:
Hello everybody.
If you want to use a 3MS or BoxTurtle mmu, then there is a need for a board with a sufficient number of stepper motor drivers. Thus: how many colors are there - so many stepper motors and, accordingly, stepper motor drivers. At the same time, each stepper motor requires at least three free pins of the microcontroller. If TMC drivers are used, then four pins are already required: en, step, dir, uart. It turns out that if I want to build myself a 3MS mmu with twelve colors, then I need to use two boards with support for six drivers. This sounds unreasonably expensive and inefficient, because usually half of the legs on the board are used to control the heater, fans, etc.

I had the idea of using demultiplexers to reduce the number of microcontroller legs used. As an example: to create an mmu like 3MS with 16 colors, only 7 microcontroller legs are required without using TMC drivers and 8 legs if TMC is used. It turns out that an additional board may not be useful for assembling a 16-color mmu if you use a demultiplexer. 

What is a demultiplexer? This is a digital logic device. It allows you to redirect the outgoing signal to the desired address. This very address is set in binary code. An illustration of the work is below. Thus, for the driver to work, three or four demultiplexers are needed, the address inputs of which will be connected together.  An approximate scheme is shown in the figure.

In this configuration, the system was tested, and for a fairly long time (about 10 days of continuous operation). There were no failures. In fact, this is an intermediate option between ERCF and 3MS. Each coil has a drive and a driver, but Klipper thinks that the motor is the same, because the en, step, dir signals for all motors come from the same pins. The disadvantage is the impossibility of simultaneous operation of several drives, but this is not required. The limit switches are connected individually and they will still need a number of pins equal to the number of limit switches.  I have a suggestion to add a function to work with demultiplexers in Happy Hare, of course, if possible. In fact, all that is required is to switch Switch0-Switch3 when changing the color. Example:

[output_pin switch_0_demultiplexor]
pin: EAZY:gpio3

[output_pin switch_1_demultiplexor]
pin: EAZY:gpio2

[output_pin switch_2_demultiplexor]
pin: EAZY:gpio1

[output_pin switch_3_demultiplexor]
pin: EAZY:gpio0

[gcode_macro Demultiplexor_chanel_0]
gcode = 
    SET_PIN PIN=switch_0_demultiplexor VALUE=0
    SET_PIN PIN=switch_1_demultiplexor VALUE=0
    SET_PIN PIN=switch_2_demultiplexor VALUE=0
    SET_PIN PIN=switch_3_demultiplexor VALUE=0

[gcode_macro Demultiplexor_chanel_1]
gcode = 
    SET_PIN PIN=switch_0_demultiplexor VALUE=1
    SET_PIN PIN=switch_1_demultiplexor VALUE=0
    SET_PIN PIN=switch_2_demultiplexor VALUE=0
    SET_PIN PIN=switch_3_demultiplexor VALUE=0

[gcode_macro Demultiplexor_chanel_15]
gcode = 
    SET_PIN PIN=switch_0_demultiplexor VALUE=1
    SET_PIN PIN=switch_1_demultiplexor VALUE=1
    SET_PIN PIN=switch_2_demultiplexor VALUE=1
    SET_PIN PIN=switch_3_demultiplexor VALUE=1

[image]
[image]

3DCoded, 2024-11-12:
> the system was tested, and for a fairly long time (about 10 days of continuous operation)

Wow! That's a lot of testing.

This looks like an effective way to add several steppers using only a couple of stepper ports. If I'm understanding correctly, you can select which motor is rotated by altering the state of a few digital pins, so T0 would be 0000, T1 0001, T2 0010, etc.

The one thing I don't like about this setup is that you need a stepper driver for each stepper. Would it be possible to use a single stepper driver and then use four demultiplexers for the four wires of the motor.

I think this system would work similarly to an array of relays (which has been in the back of my mind for a while). 

Happy Hare recently added selector types, so it shouldn't be too hard to define a custom selector class. See [mmu_selector.py](https://github.com/moggieuk/Happy-Hare/blob/v3beta/extras/mmu/mmu_selector.py).

moggieuk, 2024-11-12:
You are obviously thinking this through... have a look a `mmu_selector.py`.  The design is to allow for a "pluggable" selector implementation with the classic "LinearSelector" and the "VirtualSelector" which is the selection of multiple filament drive "gears".  I know of another in progress for 3DChamelon type design too..

Technically, `mmu_machine.py` might need som update to allow for the extra configuration, but everything thing you have described should be doable with "pluggable" extensions..

porar-oss, 2024-11-13:
> > система была протестирована и в течение довольно длительного времени (около 10 дней непрерывной работы)
> 
> Вау! Это большое тестирование.
> 
> Похоже, это эффективный способ добавить несколько шаговых двигателей, используя всего пару портов для шаговых двигателей. Если я правильно понимаю, вы можете выбрать, какой двигатель будет вращаться, изменив состояние нескольких цифровых контактов, так что T0 будет 0000, T1 0001, T2 0010 и т. д.
> 
> Единственное, что мне не нравится в этой схеме, — это то, что для каждого шагового двигателя нужен драйвер. Можно ли использовать один драйвер шагового двигателя, а затем четыре демультиплексора для четырёх проводов двигателя?
> 
> Я думаю, что эта система будет работать так же, как набор реле (о котором я уже давно подумываю). Для системы реле потребуется больше оборудования, но у меня есть несколько штук.
> 
> В Happy Hare недавно были добавлены типы селекторов, так что определить собственный класс селектора не составит труда. Смотрите [mmu_selector.py](https://github.com/moggieuk/Happy-Hare/blob/v3beta/extras/mmu/mmu_selector.py).
> 
> В настоящее время я занят переносом 3MS на HH, обновлением документации и т. д., но, возможно, смогу поработать над реализацией этого...

porar-oss, 2024-11-13:
> Would it be possible to use a single stepper driver and then use four demultiplexers for the four wires of the motor?
Indeed, the idea is good, but not without implementation problems. A demultiplexer usually acts as a logic signal switch. This usually means that it works with 5v, 3.3v or 1.65v signals. Another limitation is the internal resistance of the communication channel, which can reach 100 ohms or more. As an example, Nema 17 4401 has a winding resistance of about 1.5 ohms. This is a difference of two orders of magnitude. This directly limits the current that the chip can switch. The Nema 17 stepper motor requires a current in the range of 300-1200mA and this is when powered from 12/24v. But you can add power transistors that will be opened by a demultiplexer. Or add conventional electromagnetic relays. The figure shows a possible implementation, theoretical. At least two relays are required to switch each motor. The relay is controlled via the I2C bus. 16 relays will allow you to connect 8 stepper motors. Thus, one driver for eight engines will be used. Again, theoretically, such a kit will not require changes to the standard printer control board. Only two additional control pins from the microcontroller for the I2C bus and +5v and GRN power supply.
[image]

porar-oss, 2024-11-13:
> При аналоговом переключении нужно следить за тем, чтобы оно не нарушало состояние TMC. Не знаю, так ли это, но я бы обратил на это внимание.
 
In fact, switching by the uart pin demultiplexer has not been tested. I tested everything on a simple A4988. It is quite expensive to use 12 TMC 2209 type drivers.

porar-oss, 2024-11-13:
The idea with electromagnetic relays stopped me only by the fact that I need to connect the relay control controller to the Klipper via the I2C bus. I honestly didn't try. If this can be done, then you will need one driver and a relay board. If 9 or more coil unwinding devices are used, then several boards with relays will need to be combined into a system. I2C addresses for XL9535 can be set via the inputs of the chip itself. That is, you can use several boards in turn, changing the I2C bus address using the input combination. It turns out that there are several relay boards, and the address at which communication takes place is one. Or it is better not to change the address and just connect several XL9535 with different addresses in parallel. Except that I2C communication must be reliable so that it is not interrupted, otherwise the engine will simply shut down. There is something to think about) I can try to test the operation of the relay board, but I need to resolve the issue of Klipper communicating with it via I2C.

3DCoded, 2024-11-13:
> At least two relays are required to switch each motor.

I was thinking about a slightly different approach, where four relays controlled each motor, one for each wire, so each stepper wouldn't need to be wired to another one. 

> The relay is controlled via the I2C bus.

Does each relay need to be controlled separately, or could each "row" of relays be controlled simultaneously? I put together a (slightly messy) demo in Tinkercad, and it seems like all the relays in a "row" can be controlled with the same pin. In the video, I'm using switches but it could easily be controlled programatically.

https://github.com/user-attachments/assets/73ea171a-16b6-4a90-9e98-46182b25df58

Now that I think about it, maybe MOSFETs could be used, as they don't have that annoying clicking sound relays do. 

---

I put together some basic testing code in my fork of HH. It probably isn't printable with yet, but it does:
- implement a new `DigitalSelector` type
- add two parameters to `mmu_parameters.cfg`
    - `select_tool_macro` The macro to call when selecting tools
    - `select_tool_num_switches` The number of "rows" of relays (or MOSFETs)

The `select_tool_macro` is run when you run `MMU_SELECT GATE=xxx` (or `TOOL=xxx`). 

The macro is passed the `GATE` as a parameter, and it in binary form. Examples:

`MMU_SELECT GATE=0` passes `GATE=0 S0=0 S1=0` to the macro.
`MMU_SELECT GATE=0` passes `GATE=0 S0=0 S1=0` to the macro.

3DCoded, 2024-11-13:
The `DigitalSelector` should work with demultiplexers and relays the same way. I'm still looking into MOSFETs.

3DCoded, 2024-11-13:
I've done some more research and may have found another method: octocouplers. Here's a quick comparison of relays vs. octocouplers (for 16 motors):
| | Relay | Octocoupler |
| - | - | - |
| Quantity | 60 | 64 |
| Total price (Amazon) | $60 | $16 |
| Total price (JLCPCB) | $21 | $4 |
| Required pins | 4 | 16 |

The octocouplers would require one pin per stepper (I think). I'm wondering if I2C could help or if it would be too complex to configure in Klipper.

3DCoded, 2024-11-14:
I've renamed the `DigitalSelector` to `MacroSelector`. The same parameters work, but it now supports optocoupler-style designs and demultiplexer/relay-style designs. 

Example demultiplexer/relay-style SELECT_TOOL macro:
```
[gcode_macro SELECT_TOOL]
gcode:
    SET_PIN PIN=d0 VALUE={params.S0}
    SET_PIN PIN=d1 VALUE={params.S1}
    SET_PIN PIN=d2 VALUE={params.S2}
```

Example optocoupler-style SELECT_TOOL macro:
```
[gcode_macro SELECT_TOOL]
gcode:
    SET_PIN PIN=o{printer.mmu.gate} VALUE=0
    SET_PIN PIN=o{params.GATE} VALUE=1
```

All this is in the v3beta branch of my fork of HH [here](https://github.com/3DCoded/Happy-Hare/tree/v3beta).

I probably won't be able to test the hardware side of this, as my hardware is working fine as-is, and I prefer fewer colors. 

@porar-oss I would appreciate it if you could test this and let me know what works/what doesn't work.

porar-oss, 2024-11-14:
@porar-oss Я был бы признателен, если бы вы протестировали это и сообщили мне, что работает, а что нет.

I've finished working with the hardware right now and I'm configuring the files again. I will try to tell you about all the nuances when setting up here. Thank you so much for your help with the code! I will try to figure it out diligently and let you know within a few days.

porar-oss, 2024-11-15:
@3DCoded

It has only now been possible to make the configuration work. The error Internal error appeared during connect: 'Mmu' object has no attribute 'select_tool_num_switches'.

I had to get into selector.py Changes in the screenshot.

I just commented out the verification lines. After that, everything worked.

I'm continuing testing.

[image]

3DCoded, 2024-11-15:
@porar-oss 

I think I have fixed the error. Can you check if updating fixes it?

3DCoded, 2024-11-15:
Ok. I've tested more on the software side, and NOW it seems to be working. Can you let me know if this update works?

porar-oss, 2024-11-18:
Replaced the files `mmu_selector.py` , `mmu_sensor_manager.py` , `mmu_sequence.cfg`. The error changed to Option `'select_tool_macro'` in section 'mmu' must be specified.

[image]

3DCoded, 2024-11-18:
> `mmu_sequence.cfg`

You shouldn't need to update that.

---

In your `mmu_parameters.cfg`, add the following parameters (and edit) below the `[mmu]` line:

```
select_tool_macro: SELECT_TOOL
select_tool_num_switches: <num of demultiplexer switches>
```

porar-oss, 2024-11-18:
Adding the line `select_tool_macro: SELECT_TOOL` really helped. The string `select_tool_num_switches: <num of demultiplexer switches>` was already present. Now Happy Hare initializes without any problems. Next, I'm trying to start calibrating the mmu. The first thing I do is run `MMU_SELECT GATE=0`. After that, the error appears: in the picture below.
[image]

3DCoded, 2024-11-18:
> The string `select_tool_num_switches: <num of demultiplexer switches>` was already present.

Did you replace `<num of demultiplexer switches>` with the number of demultiplexer pins (in your first post it looked like `4`)?

---

For the internal error, can you please upload a `klippy.log`?

porar-oss, 2024-11-18:
> Did you replace <num of demultiplexer switches> with the number of demultiplexer pins (in your first post it looked like 4)?

Definitely so.

porar-oss, 2024-11-18:
Here are the logs.
[klippy.log](https://github.com/user-attachments/files/17799867/klippy.l

porar-oss, 2024-11-18:
> Ok. I've tested more on the software side, and NOW it seems to be working. Can you let me know if this update works?

I will test it whenever possible. Unfortunately, I have a lot of side work.

3DCoded, 2024-11-18:
Did you update the code and uncomment what you commented?

porar-oss, 2024-11-18:
> Did you update the code and uncomment what you commented?

Reinstalling HH helped. I'm continuing testing.

porar-oss, 2024-11-20:
Greetings. There are many additional problems during testing. Firstly, because of the non-standard location of the blade in the printhead... It's hard with this and so far I've given up the cutter. For the HH test with demultiplexers.

The MMU_CHANGE_TOOL error consistently occurs . This happens after the filament has passed the gate endstop after loading. The log file is attached.

As far as I understand, in the `MacroSelector` class, the f`ilament_release` method requires a `measure` argument.

[klippy.log](https://github.com/user-attachments/files/17831097/klippy.log)

3DCoded, 2024-11-20:
Hi @porar-oss ,

Sorry again for the errors. I have made the `filament_release` function accept an optional `measure` argument. Can you please update and let me know if it works?

porar-oss, 2024-11-23:
Greetings @3DCoded. There is some inconvenience: after restarting the Klipper, the address of the selected gate does not arrive on the demultiplexers.

I.e., first you need to enable the desired gate address with a command from Mainsail, and only after that start printing.

If you then select another tool, then everything works normally: the demultiplexers switch. The inconvenience is only after restarting.

A linear selector cannot have such a problem. If the filament is already loaded, the selector will not move.

In our case, it is necessary to programmatically issue a signal during initialization to select the last loaded gate.

3DCoded, 2024-11-23:
Hi @porar-oss,

I have added an experimental fix for this. Can you please update and let me know if that works? Thank you.

porar-oss, 2024-11-23:
I checked it out. It really works. If there is anything additional, I will definitely write.
Thanks for the implementation @3DCoded!

3DCoded, 2024-11-23:
@porar-oss 

After you do some more tests, let me know, and I can submit a PR.

porar-oss, 2024-12-01:
Good afternoon @3DCoded.

I am still continuing testing. Everything related to demultiplexers works normally. MMU drivers operate in STEP/DIR mode.

No problems have been identified so far. Current achievement: the tool has been changed 88 times in a row without errors. It's not much, but I think there will be more)

[image]

3DCoded, 2024-12-01:
That's great so far! I'm thinking maybe 100 swaps would be a good threshold for reliability.

3DCoded, 2024-12-09:
I submitted a PR: #545

3DCoded, 2024-12-14:
@porar-oss ,

PR #545  is merged with this implementation!

3DCoded, 2024-12-14:
I would close this now.

moggieuk, 2026-10-21:
Closing this, since it was implemented as the `MacroSelector` in PR #545 (thanks @3DCoded and for all the testing!) and it's carried forward into v4.

In v4, choose "MacroSelector" as the "Selector type?" in menuconfig (this choice is offered when you configure a custom MMU type), then in `mmu_parameters.cfg` set:
- `select_tool_macro`: the name of your macro, e.g. `SELECT_TOOL`
- `select_tool_num_switches`: the number of demultiplexer address lines, e.g. `4` for 16 gates

On every gate change HH calls your macro with `GATE=`, `LGATE=` (the gate local to that unit), and `S0=`, `S1=`, ... set to the binary address bits, so the macro only has to `SET_PIN` each line.

If there's something v4 still doesn't do for you, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk