moggieuk/Happy-Hare #514: Docu update Slicer settings

dominikacri, 2024-11-13:
Hi,
maybe a addition to the slicer setup for prusa slicer. 
Its also needed to set the ramming time to 0 in ramming setting for each filament. 

Otherwise the E-15.000 is still in the gcode output and leads in my case to extruder skips.

;WIDTH:0.5
G1 E-15.0000
G1 E10.4965
G1 E2.9990
G1 E1.4995
G4 S0

Described also here:

https://github.com/prusa3d/PrusaSlicer/issues/13175

dominikacri, 2024-11-13:
Correction, its now needed to set cooling tube length exactly to 0 to remove the E-15.000

WipeTower.cpp prusa slicer

if (m_semm && (m_cooling_tube_retraction != 0 || m_cooling_tube_length != 0)) {
        float total_retraction_distance = m_cooling_tube_retraction + m_cooling_tube_length/2.f - 15.f; // the 15mm is reserved for the first part after ramming
        writer.suppress_preview()
              .retract(15.f, m_filpar[m_current_tool].unloading_speed_start * 60.f) // feedrate 5000mm/min = 83mm/s
              .retract(0.70f * total_retraction_distance, 1.0f * m_filpar[m_current_tool].unloading_speed * 60.f)
              .retract(0.20f * total_retraction_distance, 0.5f * m_filpar[m_current_tool].unloading_speed * 60.f)
              .retract(0.10f * total_retraction_distance, 0.3f * m_filpar[m_current_tool].unloading_speed * 60.f)
              .resume_preview();

moggieuk, 2024-11-16:
Doc updated. Thanks.

closed 2024-11-16 by moggieuk