moggieuk/Happy-Hare #517: In v3 beta, EREC_CUTTER_ACTION performs unnecessary gear move when VARIABLE_CUT_ATTEMPTS set to 1

iamnotahippy, 2024-11-15:
The changes to the EREC cutter macros in the v3 beta are helpful, but the 1 mm retraction after the cut is unnecessary if we only want to do a single cut attempt. If the variable is set to 1, the cutting sequence is CUTTER OPEN > SERVO DOWN > FEED > SERVO UP > SERVO DOWN > RETRACT >  CUTTER CLOSE > UNLOAD GATE > SERVO UP

This causes excess wear on our already fragile servo.

moggieuk, 2024-11-15:
So am I understand two issues here (btw I don't use EREC myself so I don't have first hand experience):
1. The logic in the cutter macro doesn't need to retract 1mm, if only one cut is being made.
2. The "servo" movement is referring to the filament selector servo I assume and that is oscillating more that necessary..?  Can you confirm #2.

iamnotahippy, 2024-11-16:
Sorry I wasn't as clear as I could have been. My understanding is that for cutting attempts > 1, the 1mm retract is to prevent the cutter from catching on the protruding filament when opening again for the second cut.  When only performing one cut, there's only one cycle of the cutter, so this move isn't necessary and results in 2 back-to-back _MMU_STEP_MOVE commands, one of which isn't necessary. 

Here are some videos to illustrate:

Current behavior with variable_cut_attempts = 1

https://www.youtube.com/watch?v=PqH7cpCikxY

Current behavior with variable_cut_attempts = 2

https://www.youtube.com/watch?v=pgDIQqsivx4

Desired behavior with variable_cut_attempts = 1

https://www.youtube.com/watch?v=soQp2wxB8v0

iamnotahippy, 2024-12-05:
Fixed in most recent update

closed 2024-12-05 by iamnotahippy