moggieuk/Happy-Hare #530: Internal error during connect: 'ManualExtruderStepper' object has no attribute 'cmd_SET_E_STEP_DISTANCE'

koekie40, 2024-11-20:
I am going through set up now and I'm getting this error 'ManualExtruderStepper' object has no attribute 'get_endstop_names'. 
I am still kind of new Happy Hare and the ERCF. I am following the guide and I'm afraid I'm over looking something simple.
[image]

ningpj, 2024-11-23:
how up to date is klipper?  Theres been alot of changes to klipper over the last 12-mths that required updates to HH and is now a dependency between to two.  Update klipper and try again

moggieuk, 2024-12-05:
Hi, sorry for the delay.  You are using and VERY old version of Happy Hare that is deprecated and not supported anymore.  The new version v2.7.3 and soon to be released v3.0.1 don't use ManualSteppers.

It is possible that if you did upgrade to v2.7.3 / v3.0.0 from a very old version that the upgrade of `mmu_hardware.cfg` was not able to be performed.  The best thing to do is to save the info from that file (then DELETE the file) and re-run:
```
./install.sh -i
```
This will re-create a correct `mmu_hardware.cfg` to which you can apply the configuration from your previous install.

closed 2025-01-18 by github-actions[bot]