moggieuk/Happy-Hare #531: sync_feedback_tension always triggered

Belstrekkie, 2024-11-20:
I've been using a filament [stress sensor](https://www.printables.com/model/803180-voron-ercf-filament-stress-sensor/comments/1746435) for quite a long time without any issues, but for a while now I'm getting warnings that both tension and compression are triggered at the start of a print. After that I replaced the Hall sensors, but the issue remains.
MMU_SENSORS showed my the compression sensor is working as expected, but the tension sensor is always triggered, even if the sensor is not connected to the mcu at all:
-  sync_feedback_tension: TRIGGERED

Although I think it might be a hardware issue, I found that even after disabling sync_feedback in mmu_parameters.cfg and removing the pin allocation (and even the entire line 233/234) from the mmu_hardware.cfg, the MMU_Sensors command still show after a restart of the printer:
- sync_feedback_compression: open
- sync_feedback_tension: TRIGGERED
I wonder if this is normal? Should these sensors status be shown if not configured and is triggered in this case standard for the tension sensor?

moggieuk, 2024-12-04:
The two sensors are completely independent and this does sound like a h/w issue.  Do you have the pull up resister `^` set correctly?  Or perhaps you need to invert one of the sensors with `!`?

Note that the best way to disable a sensor is to remove the pin assignment in `[mmu_sensors]` section or commenting out that line.  If `MMU_SENSORS` still reports the sensor then it isn't disabled somehow.  Note that `MMU_SENSORS DETAIL=1` will show disabled sensors as well.

Perhaps share your `mmu_hardware.cfg`..

Belstrekkie, 2024-12-04:
[mmu_hardware.txt](https://github.com/user-attachments/files/18011547/mmu_hardware.txt)
Here is the config file (changed to txt as I can't share cfg here). Uncommentend and no pins configured for both sensors, but still mmu_sensors reports their status as triggered for the tension and open for compression...

moggieuk, 2024-12-08:
Firstly, the default wildcard include line `[include mmu/base/*.cfg]` will pick up every cfg in the directory.  Klipper allows duplicate config in some cases and whichever is loaded last (generally alphabetically) will be in effect.  Glad you found this because it was confusing me.

As far as pin config goes. It is possible that your MMB board has a problem, but PA3 and PA4 pins are shared with the DIAG for motors M0 and M1.  The jumper selection on the board defines how they are used.    The klipper `!` simply inverts the signal and will not alter behavor.  The `^` is usually required for sensor and activates a software controlled pull-up resistor (not setting this often causes the pin to oscillate).

Assuming PA3 is not set up for DIAG (jumper inserted) and you are sure wiring is correct (you can always switch PA3 and PA4 for double check this), then this GPIO pin on the MMB board may have gone bad.  You can use other pins though .. are they all currently in use?

Here is my reference to MMU board layout: 
https://github.com/moggieuk/Happy-Hare/wiki/Mcu-Reference#bigtreetech-btt-mmb-can-v1

Belstrekkie, 2024-12-08:
Thank you for the information, that solved my issue. I wasn't aware that PA3/PA4 were shared with the diagnostic pins, as this is mentioned in the pin layout of the MMB. Still don't understand why it worked well for months, but now I solved the issues by moving to the sensor to the I2C pins and this seems to work with the sensors behaving as expected.

Thanks!

moggieuk, 2024-12-15:
You are welcome!   Closing issue.

closed 2024-12-15 by moggieuk