moggieuk/Happy-Hare #533: Motor sync issue

Darkchild971, 2024-11-21:
My printer is equipped with a belay. I followed the wiki for its configuration and the correct information is displayed by the mmu_sensor command

(triggered when I compress the belay by hand)

config extract:
sync_feedback_tension_pin:: ^mmu:gpio3
sync_feedback_compression_pin: 

nevertheless even with an extreme value forsync_multiplier 

example : 
sync_multiplier_high: 1.2
sync_multiplier_low: 0.8

The filament is never under tension so the switch is never activated as if the sync_multiplier_low parameter was not taken into account.
I tried to get more information with log level: 2 but no information is displayed even when I manually force the activation of the switch during printing

Note: I am sure that I do not have a difference of 20% between my extruder and the mmu_gear, both were calibrated with a caliper

Darkchild971, 2024-11-23:
thanks for the feedback. I use a tradrack so the problem does not come from that side for me

porar-oss, 2024-12-01:
Faced with a similar problem.

During printing, the rotation distance parameter does not change when the compression or tension sensors are triggered. Unfortunately, before that they worked during the first printing, and then they stopped. Then they started working again, but inverted.

The very fact of pressing the Happy Hare sensor determines, but does not change the coefficient.

moggieuk, 2024-12-04:
Thanks. I'll take a look at this since there are a few of you reporting but I'd need `mmu_log` to diagnose specific problems.

One this you might like to try is the `v3beta` branch.  This will be pushed to `main` very soon.  It would be good to know if the problem still exists on this release because this is where I would be addressing any problems.

Reminder. To switch to a specific branch:
```
./install.sh -b v3beta
```

To switch back to "main" when released:
```
./install.sh -b v3beta
```

Darkchild971, 2024-12-04:
[mmu.log](https://github.com/user-attachments/files/18014106/mmu.log)
Here is my log

Darkchild971, 2024-12-05:
After more testing:
The belay operates normally for the first 5-10 minutes and then it is no longer controlled (it remains in the compressed position opposite the sensor)

To succeed in making it work I need at least (the problem is the same by increasing these values)

sync_multiplier_high: 1.15
sync_multiplier_low: 0.85

Darkchild971, 2024-12-05:
I just did the test with a turtleneck (2 sensors).
The behavior is the same: the tension sensor is never reached and little by little the belay moves less and less away from the compression sensor until it no longer leaves it until the end of printing

moggieuk, 2024-12-08:
Can you send a `mmu.log` with log_file_level: 3 (I need the TRACE output).  You can do this temporarily with:
```
MMU_TEST_CONFIG log_file_level=3
```

I just ran a test and it worked as expected.  Here is some (simplified) log output where I annotate what is happening...
```
This is at the start of the print. Neither tension or compression sensor is triggered. Here the gear stepper is synced at the default calibrated rotation distance:

14:06:00     TRACE: New sync direction: extrude
14:06:00     TRACE: Updated sync multiplier: 1.0000
14:06:00     TRACE: Setting gear motor rotation distance: 22.731687

Later the tension triggers:
14:06:04     TRACE: Got sync force feedback update. State: -1
14:06:04     TRACE: Updated sync multiplier: 1.0500
14:06:04     TRACE: Setting gear motor rotation distance: 21.649226
The rotation distance is smaller, meaning HH will turn the gear stepper faster to relieve the tension

The tension sensor un-triggers (and normal speed is restored):
14:06:07     TRACE: Got sync force feedback update. State: 0
14:06:07     TRACE: Updated sync multiplier: 1.0000
14:06:07     TRACE: Setting gear motor rotation distance: 22.731687

Later the compression triggers:
14:06:10     TRACE: Got sync force feedback update. State: 1
14:06:10     TRACE: Updated sync multiplier: 0.9500
14:06:10     TRACE: Setting gear motor rotation distance: 23.928091
The rotation distance is larger, meaning HH will turn the gear stepper slower to relieve the compression

The tension sensor un-triggers (and normal speed is restored):
14:06:12     TRACE: Got sync force feedback update. State: 0
14:06:12     TRACE: Updated sync multiplier: 1.0000
14:06:12     TRACE: Setting gear motor rotation distance: 22.731687
```

Maybe you can help and review your `mmu.logs` and see if you can detect what is (or not happening).
Note that in a real print you might have retraction sufficient to cause a rotation distance change opposite to what is shown here. I.e. compression must be treated like tension if retracting...

moggieuk, 2024-12-15:
You're welcome. Closing issue.

closed 2024-12-15 by moggieuk