moggieuk/Happy-Hare #537: Different speeds for loading and unloading the filament.

porar-oss, 2024-11-23:
Greetings.

During testing, the mmu type B found that in the case of loading, the speed of movement of the filament is limited by the mass of the coil with additional resistance from the Bowden tube. In the case of unloading, the resistance of the coil remains the same, and the resistance of the Bowden tube becomes less.

The specific speeds of 265mm/s are loading, 300mm/s unloading (these are the speeds at which the mmu worked stably). The data were obtained using nema17 17HS4401 motors, supply voltage 24v, current 0.8A. BMG gear, with a 20:16 step-up gearbox. The length of the bowden is 1500mm, the inner diameter of the tube is 2.5mm.

During the configuration, Happy Hare found that it is not possible to simply set different speeds for loading/unloading the filament. I think such functionality can fine-tune the mmu in order to save time changing plastic.

P.S. for speeds of 265 and 300mm/s (for loading and unloading, respectively), the time gain is about 6%, compared with a speed of 265mm/s for both operations.

moggieuk, 2024-12-05:
There are different speeds for loading from spool vs from buffer, but this was inspired from MMU designs with non-powered buffering systems. In this setup the unload is the same as the "from buffer" load which seems appropriate.

In the case of adding powered rewinder this distinction isn't as clear and really every load is from the spool.  Perhaps that is the best fix... if rewinder then never mark filament as "from buffer" so:
```
gear_from_buffer_speed
```
Would be applied to unloads only
```
gear_from_spool_speed
```
Would be applied to loads

Maybe not the best naming but I acknowledge the distinction..

igiannakas, 2024-12-07:
Btw this should be the same for a filamentalist set up. No concept of buffer speeds there too :)

alsonif you’re considering opening up this area it may be worth considering a by lane speed setting. Why? Because for example in a filamentalist setup you may have 6 spools in the filamentalist enclosure that are near the ERCF vs the last two being further away and with longer Bowden runs meaning more resistance. So the near ones can go fast and the further away ones slower. Basically instead of a single value, set this to be a comma separated list of speeds :)

moggieuk, 2025-01-19:
This has now been implement in the `mutl_mmu` branch.  That will be pushed to `main` in the next 1-2 weeks. Unload speed and accel can be set with: `gear_unload_speed` and `gear_unload_accel`
```yml
# Long moves are faster than the small ones and used for the bulk of the bowden movement. You can set two fast load speeds
# depending on whether pulling from the spool or filament buffer (if fitted and not the first time load). This can be helpful
# in allowing faster loading from buffer and slower when pulling from the spool because of the additional friction (prevents
# loosing steps). Unloading speed can be tuning if you have a rewinder system that imposes additional limits.
# NOTE: Encoder cannot keep up much above 450mm/s so make sure 'bowden_apply_correction' is off at very high speeds!
#
gear_from_spool_speed: 80               # mm/s Speed when loading from the spool (for the first time if has_filament_buffer: 1)
gear_from_spool_accel: 100              # Acceleration when loading from spool
gear_from_buffer_speed: 150             # mm/s Speed when loading filament from buffer. Conservative is 100mm/s, Max around 400mm/s
gear_from_buffer_accel: 400             # Normal acceleration when loading filament
gear_unload_speed: 80                   # mm/s Use (lower) speed when unloading filament (defaults to "from spool" speed)
gear_unload_accel: 100                  # Acceleration when unloading filament (defaults to "from spool" accel)
```

Related is a new way to specify if you have a filament buffer.  This effects status display but also the "from_buffer" speed above.
```
has_filament_buffer: [0|1]
```
in `mmu_parameters.cfg`

closed 2025-03-04 by github-actions[bot]

moggieuk, 2026-10-23:
For reference: related follow-up work on heavy/full-spool handling is now being worked on in #856 (spool weight from Spoolman scaling espooler power).

moggieuk, 2026-10-23:
For reference: the related heavy/full-spool work (#856) has been merged into `development` (#1450) and will be released in **v4.2.0**.