moggieuk/Happy-Hare #538: Manage Accelerations during MMU sequence commands

igiannakas, 2024-11-26:
Unfortunately HH currently I dont believe manages accelerations for any of the moves in its "domain". What this means is that any travel, cut, restore move is done with the last print acceleration. 

Typically this will be an external perimeter or the acceleration of the sparse infill, which is not really an issue; however for any tool changes happening on the first layer, the moves behave like they are in slow motion. Equally depending on the print, the last move acceleration may be a top surface or other area where lower accelerations are used.

It would be good if HH captured the current print acceleration at the start of the MMU sequence, exposed a variable for the user to control move accelerations, set the user requested value and restored the original print acceleration at the end of an MMU operation.

This can be done in the _MMU_PRE_UNLOAD and _MMU_PRE_LOAD macros and set to the saved acceleration in the _MMU_POST_UNLOAD and _MMU_POST_LOAD macros (in all four places to cover the scenario of simply a load move, like when starting a print or just an unload move when ending a print).

Code for this could look like:

```
    {% set original_accel = printer.toolhead.max_accel %}
    {% set original_minimum_cruise_ratio = printer.toolhead.minimum_cruise_ratio %}
    SET_VELOCITY_LIMIT ACCEL=10000 MINIMUM_CRUISE_RATIO=0.1
```

and 

`SET_VELOCITY_LIMIT ACCEL={original_accel} MINIMUM_CRUISE_RATIO={original_minimum_cruise_ratio}`

moggieuk, 2024-12-04:
That is a very good point and you are correct (I think).   Until all the parking macros the movement was minimal but I agree that the parking logic should allow for controlling this. 

It certainly could be done in pre/post macros but it is difficult (impossible with current structure) to guarantee restoration. I.e. the post macro may never be run on failure.  It would however but quite easy for HH to set and restore using a "context manager" or try/finally wrapper to guarantee state.

Let me think about a good way to integrate this..

moggieuk, 2024-12-05:
Done.  Currently only in "development" branch. Will be in "v3beta" later today...
```
# When Happy Hare calls out to a macro for user customization and for parking moves these settings are applied and the previous
# values automatically restored afterwards. This allows for deterministic movement speed regardless of the starting state.
#
macro_toolhead_max_accel: 0             # Default printer toolhead accelaration applied when macros are run. 0 = use printer max
macro_toolhead_min_cruise_ratio: 0.5    # Default printer cruise ratio applied when macros are run
```

closed 2024-12-05 by moggieuk

igiannakas, 2024-12-05:
Thank you very much for this!!

I’ll give it a test and in general the v3 version at some point over Christmas :)