moggieuk/Happy-Hare #543: Could not exclusively lock port /dev/ttyACM1

DuckersMcQuack, 2024-12-02:
Waited months for parts and "energy" to continue, All parts are installed and ready to go, but now it can't seem to read from the mmu anymore. Or "lock in"

Using siboor kit with MMB CAN, connected to pi with usb C, hooked up with SKR 1.3 to a pi 4 8GB

mcu 'mmu': Starting serial connect
mcu 'mmu': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/ttyACM1: [Errno 11] Resource temporarily unavailable
mcu 'mmu': Unable to open serial port: [Errno 11] Could not exclusively lock port /dev/ttyACM1: [Errno 11] Resource temporarily unavailable

moggieuk, 2024-12-05:
This isn't really a Happy Hare issue but either a config issue (are you sure `/dev/ttyACM1` is the connection to your mcu?).  Or a temporary problem that could be solved by rebooting your rpi.

I suspect the former because /dev/ttyACM1 is usually an alias and it is better to use the correct name for the connection.  Use:
```
ls /dev/serial/by-id/*
```
To list the possible devices.
It will likely look something like:
```
/dev/serial/by-id/usb-Klipper_rp2040_E66160F42346B637-if00
```
(your specific id will be different)

DuckersMcQuack, 2024-12-06:
It appears it was all because i didn't realized i shouldn't use the regular power terminals, but the big black cable that came with the ERCF kit. And i'm now also almost complete with the tuning. Just rotation distance remaining.

closed 2024-12-06 by DuckersMcQuack