moggieuk/Happy-Hare #548: Mirrored Filament Blocks

SamueldaCostaAraujoNunes, 2024-12-10:
I noticed that, like me, many people use their ERCFs backwards, with the front pointing towards the back of the printer. This allows them to, for example, place filaments on top of the printer. The bad thing is that there is no possibility today of inverting the doors. So that, when you see the front of the printer, the sequence of extruders would be from left to right. Is this feasible?

moggieuk, 2024-12-14:
I'd certainly entertain a PR for this but it would be very low priority for me to work on... just too much testing.

It's probably not a complex change.  Any reference to `self.gate_selected` and `self.tool_selected` would need to be changed to call function that applied the inversion. But there would be a lot of places to test.

SamueldaCostaAraujoNunes, 2024-12-15:
Yes, I would probably make a parameter of a macro, inverted gates. And when I was going to get the position of the tool or the gate I would do, NUMBER OF GATES - CURRENT GATE

I'm going to fork the project and test it here

SamueldaCostaAraujoNunes, 2024-12-18:
I found an existing solution that worked very well:
https://github.com/moggieuk/Happy-Hare/wiki/Tool-and-Gate-Maps#:~:text=MMU_REMAP_TTG%20MAP%3D8%2C7%2C6%2C5%2C4%2C3%2C2%2C1%2C0

moggieuk, 2026-10-21:
Closing, since as you found, the tool-to-gate map already handles a reversed MMU.

In v4 it's `MMU_TTG_MAP MAP=8,7,6,5,4,3,2,1,0` (`MMU_REMAP_TTG` still works as an alias), so T0 uses the right-hand gate as you face the printer, and so on. The map is saved and survives restarts. If you'd like it to be your baseline, also set `default_ttg_map: 8, 7, 6, 5, 4, 3, 2, 1, 0` in `mmu.cfg` so `MMU_TTG_MAP RESET=1` restores it.

If there's something v4 still doesn't do for you, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk