moggieuk/Happy-Hare #552: Error After Eject

barsamgr, 2024-12-12:
Hello,

After updating to the latest version (3.0.1.11) I am getting an error from MMU after the printing is finished and after the EJECT process putting the printer on PAUSE. Following is the Log.

21:29:26 > MMU_EJECT
21:29:26   DEBUG: Saving toolhead gcode state and position (X:175.3 Y:150.1 Z:59.8 E:2918.0) for unload
21:29:26 Saving toolhead position (x:175.3, y:150.1, z:59.8)
21:29:26 Retracting 2.0mm
21:29:26 Parking toolhead at (x:20.0, y:357.5, z:59.8) for unload operation
21:29:27   DEBUG: Unloading tool T0
21:29:30 Unloading filament...
21:29:30 [T0] < En <<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
21:29:31 Unsynced MMU from extruder (sync feedback deactivated)
21:29:31 Restoring MMU gear stepper run current to 100% configured (1.20A)
21:29:31   DEBUG: Setting pressure advance for tip forming: 0.000000
21:29:31 Forming tip...
21:29:31 Retracting filament 19.0mm prior to cut
21:29:33 > MMU_STATS COUNTER=cutter_blade INCR=1
21:29:33 Pushing filament fragment back 5.0mm after cut
21:29:36   DEBUG: Restoring pressure advance: 0.042000
21:29:36 [T0] < En <<<<<<<<< (*) << [Ex << (*) <. Nz] -34.0mm (e:33.6mm)
21:29:36   DEBUG: Extracting filament from extruder
21:29:36   DEBUG: Reverse homing up to 92.1mm to extruder sensor (synced) to exit extruder
21:29:40 [T0] < En <<<<<<<<<|(e) .. [Ex .. (t) .. Nz] -103.3mm (e:101.8mm)
21:29:40   DEBUG: Filament should be out of extruder
21:29:40   DEBUG: Unloading bowden tube
21:29:40   DEBUG: Sensors: {'mmu_pre_gate_0': True}
21:29:47 Warning: Excess slippage was detected in bowden tube unload. Gear moved 931.8mm, Encoder measured 910.4mm
21:29:47 [T0] < En <........ (e) .. [Ex .. (t) .. Nz] -1017.9mm (e:1012.2mm)
21:29:49 [T0] < En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED -1050.2mm (e:1030.5mm)
21:29:49 Unload of -1050.2mm filament successful (encoder measured 1030.5mm)
21:29:49   DEBUG: Autotune: bowden move ratio: 0.000000, Extra homing movement: 0.0mm
21:29:50   DEBUG: Continuing from printing state after unload
21:29:50   DEBUG: Sensors: {'mmu_pre_gate_0': True, 'extruder': False, 'toolhead': False}
21:29:50 Synced MMU to extruder (sync feedback activated)
21:29:50 Modifying MMU gear stepper run current to 50% (0.60A) for extruder syncing
21:29:50 Restoring toolhead position to last pos: (x:175.3, y:150.1, z:59.8)
21:29:50   DEBUG: Ensuring correct gcode state and position (X:175.3 Y:150.1 Z:59.8 E:3165.0) after unload
21:30:00 > __MMU_SENSOR_RUNOUT SENSOR=extruder EVENTTIME=40618.292040882
21:30:00 MMU issue detected. Print will be paused
         Reason: Filament runout occured at extruder. Manual intervention is required
21:30:00 After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
21:30:00   DEBUG: Cannot save toolhead position or z-hop for pause because not homed
21:30:00   DEBUG: Job State: PRINTING -> PAUSE_LOCKED (MMU State: Encoder: Disabled, Synced: True, Paused temp: 0.0, Resume to state: printing, Position saved for: , pause_resume: False, Idle timeout: 72000.00s)
21:30:00 > PAUSE

moggieuk, 2024-12-14:
Try using `MMU_UNLOAD`.   MMU_EJECT forces the filament completely out of the MMU now -- still shouldn't throw a runout so I'll look into that..

closed 2024-12-14 by moggieuk

barsamgr, 2024-12-15:
Yes I figured that out and I had already changed to MMU_UNLOAD. Thank you for your response.