moggieuk/Happy-Hare #553: Error on toolchange after most recent update

iamnotahippy, 2024-12-13:


milkmother69, 2024-12-13:
I got the same error, fixed it by reverting to commit 271af7c.
Seems like there was a bunch of changes involving the gear stepper recently, probably a mistake in there somewhere.

Sparkss, 2024-12-14:
> I got the same error, fixed it by reverting to commit [271af7c](https://github.com/moggieuk/Happy-Hare/commit/271af7ccf8adb5ad296b891fd7cf6e6e8dae4de4). Seems like there was a bunch of changes involving the gear stepper recently, probably a mistake in there somewhere.

Same issue here. Now to try to remember how to revert to a previous commit with git :).

moggieuk, 2024-12-14:
looking...

moggieuk, 2024-12-14:
FIXED.  Just force an update with `update-manager` in Mainsail, etc. (i.e. you might have to hit the refresh button to see the update)

iamnotahippy, 2024-12-14:
Confirmed fixed.

closed 2024-12-14 by iamnotahippy